ZiwenZhuang/parkour #1: What is the compatible GPU and driver version

sunzhon, 2023-09-18:
Hi, I am trying to reproduce your great control algorithm, but I encountered a problem when I ran legged gym using this command: 'python legged_gym/scripts/train.py --headless --task a1_field'

And then I got this error:

python legged_gym/scripts/train.py --headless --task a1_field
Importing module 'gym_38' (/home/user/workspace/parkour/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/suntao/workspace/parkour/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
PyTorch version 1.10.0+cu113
Device count 1
/home/user/workspace/parkour/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/user/.cache/torch_extensions/py38_cu113 as PyTorch extensions root...
Emitting ninja build file /home/user/.cache/torch_extensions/py38_cu113/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
Setting seed: 1
Using LeggedRobotField.__init__, num_obs and num_privileged_obs will be computed instead of assigned.
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: enabled
Traceback (most recent call last):
  File "legged_gym/scripts/train.py", line 49, in <module>
    train(args)
  File "legged_gym/scripts/train.py", line 43, in train
    env, env_cfg = task_registry.make_env(name=args.task, args=args)
  File "/home/suntao/workspace/parkour/legged_gym/legged_gym/utils/task_registry.py", line 99, in make_env
    env = task_class(   cfg=env_cfg,
  File "/home/suntao/workspace/parkour/legged_gym/legged_gym/envs/base/legged_robot_field.py", line 19, in __init__
    super().__init__(cfg, sim_params, physics_engine, sim_device, headless)
  File "/home/suntao/workspace/parkour/legged_gym/legged_gym/envs/base/legged_robot.py", line 72, in __init__
    super().__init__(self.cfg, sim_params, physics_engine, sim_device, headless)
  File "/home/suntao/workspace/parkour/legged_gym/legged_gym/envs/base/base_task.py", line 84, in __init__
    self.create_sim()
  File "/home/suntao/workspace/parkour/legged_gym/legged_gym/envs/base/legged_robot.py", line 238, in create_sim
    self._create_envs()
  File "/home/suntao/workspace/parkour/legged_gym/legged_gym/envs/base/legged_robot_field.py", line 561, in _create_envs
    self.motor_strength = torch_rand_float(
RuntimeError: The following operation failed in the TorchScript interpreter.
Traceback of TorchScript (most recent call last):
RuntimeError: nvrtc: error: invalid value for --gpu-architecture (-arch)

I thought this may related to the compatibility of the GPU, IsaacGym, and GPU driver..
So I am curious which Nvidia GPU you used.

Thank you!

ZiwenZhuang, 2023-09-18:
Hi,

The GPU I used for training is Nvidia 3090 with 24GB memory or Nvidia A10 with 23.8GB memory.

From the `nvidia-smi` output on Nvidia 3090, it says `NVIDIA-SMI 525.116.04   Driver Version: 525.116.04   CUDA Version: 12.0`

From the `nvidia-smi` output on Nvidia A10, it says `NVIDIA-SMI 515.105.01   Driver Version: 515.105.01   CUDA Version: 11.7`

To be noted that `CUDA Version` in 11.4 won't work when rendering is needed.

Or, you can try to install pytorch by `pip install -q torch==1.10.0+cu111 torchvision==0.11.0+cu111 torchaudio==0.10.0 -f https://download.pytorch.org/whl/torch_stable.html`

Also, please use isaacgym preview4. I didn't test the compatibility of history versions.

Best,

closed 2023-09-23 by MarkFzp

sunzhon, 2023-09-28:
Thank you for your reply. I changed my GPU, using RTX3090 with a 470 driver and Cuda:11.4. It worked well.