ZiwenZhuang/parkour #3: Training separate skills and distillation process

xkhainguyen, 2023-10-15:
Hi,

First of all, thank you for open-sourcing the awesome work. 

I want to reproduce the policy but am unsure about the correct steps. This is my assumed procedure (I highlight things not clear):

1. Train specialized skills
- Walk (hard): 
`python legged_gym/scripts/train.py --headless --task a1_field`
- Climb (soft):
virtual_terrain = False
comment the `class sensor`
update `climb["depth"]` (I'm not sure what you meant, the correct thing may be **`climb["height"]`**?)
 `python legged_gym/scripts/train.py --headless --task a1_climb --load_run **{the dir of walk?}**`
 `python legged_gym/scripts/train.py --headless --task a1_climb --load_run Oct15_14-21-57_WalkingBase_pEnergySubsteps2e-5_aScale0.5`
- Climb (hard):
virtual_terrain = True
Uncomment the `class sensor`
update `climb["depth"]` (I'm not sure what you meant, the correct thing may be **`climb["height"]`**?)
 `python legged_gym/scripts/train.py --headless --task a1_climb --load_run **{the dir of previous soft climb?}**`
 `python legged_gym/scripts/train.py --headless --task a1_climb --load_run Oct15_14-27-56_Skillclimb_pDofLimit-1e-01_climbHeight0.2-0.6_trackBlockLen1.6`
- Leap (soft):
virtual_terrain = False
comment the `class sensor`
 `python legged_gym/scripts/train.py --headless --task a1_leap --load_run **{the dir of walk?}`**
 `python legged_gym/scripts/train.py --headless --task a1_leap --load_run Oct15_14-21-57_WalkingBase_pEnergySubsteps2e-5_aScale0.5`
- Leap (hard):
virtual_terrain = True
Uncomment the `class sensor`
 `python legged_gym/scripts/train.py --headless --task a1_leap --load_run **{the dir of previous soft leap?}**`
- Same steps for crawl and tilt

2. Distill the parkour policy
- Update the fields with my hard trained skill directories
```
sub_policy_paths = [ # must in the order of obstacle ID, Replace the folder name with your own training logdir
    os.path.join(logs_root, "field_a1/{your walking policy}"),
    os.path.join(logs_root, "field_a1/{your tilting policy}"),
    os.path.join(logs_root, "field_a1/{your crawling policy}"),
    os.path.join(logs_root, "field_a1/{your climbing policy}"),
    os.path.join(logs_root, "field_a1/{your leaping policy}"),
]
climb_down_policy_path = os.path.join(logs_root, "field_a1/{your climbing down policy}")
```
**What is `climb_down_policy_path`?**
- Run data collection **(before or after training?)**
`python legged_gym/scripts/collect.py --headless --task a1_distill` **(do I need load_run and what will it be)**?
- Update the `A1FieldDistillCfgPPO.runner.pretrain_dataset.data_dir` field with a list of dataset directories. **(Not sure how to do this?)**
Comment out the `A1FieldDistillCfgPPO.runner.pretrain_dataset.scan_dir` field.
- Run `python legged_gym/scripts/train.py --headless --task a1_distill` to start distillation **(before or after collecting?)**
- Comment out `A1FieldDistillCfgPPO.runner.pretrain_dataset.data_dir` field and uncomment `A1FieldDistillCfgPPO.runner.pretrain_dataset.scan_dir` field. (in `a1_field_distill_config.py`)
- Update the `A1FieldDistillCfgPPO.runner.load_run` field with your last distillation log.
- Run `python legged_gym/scripts/train.py --headless --task a1_distill` to start dagger. The terminal will prompt you to launch a collector process.
- To run the collector process, change `RunnerCls` to `DaggerSaver` (as in lines 20-21). Run `python legged_gym/scripts/collect.py --headless --task a1_distill --load_run {the training run prompt by the training process}`.
- Press Enter in the training process when you see some data is collected (prompted by the collector process). The training process will load the collected data and start training.

3. Visualization
`python legged_gym/scripts/play.py --task {task} --load_run **{dir of the log of corresponding task?}**` 

**What are the differences between `distill_{robot}_dagger` and `distill_{robot}`?**
**Can I run distillation in one GPU?**

Thanks

ZiwenZhuang, 2023-10-16:
Hi,

* For running the walk training, the step is correct.

* For training in soft/hard dynamics, you mistake the `virtual_terrain` option. You must set `virtual_terrain = True` to enable the soft dynamics and let the environment compute penetration depths.

* The way you set `--load_run` is correct.
    For the concept of log dir, which will be set to the `load_run` field, it is a directory that contains the training config file as well as some checkpoints. By default of each task, the loading mechanism will search in `logs/field_a1/` if you set it as the run name (e.g. `Oct15_14-21-57_WalkingBase_pEnergySubsteps2e-5_aScale0.5`). Or you can set the `load_run` as an absolute path to the log dir.

* For the `climb["depth"]` field. 
    It is the length of the virtual climbing obstacle along the x-axis, which is also the forward direction. This field is not used in non-virtual (hard) terrain. You should set it smaller (e.g. 0.1 or 0.2). Then the policy will learn to jump upward instead of struggling to learn to jump forward.

* For the sensor class in each specialized skill training.
    It is designed to simulate action delay on the real robot. You can comment this out if you find it hard to train, then uncomment it to fine-tune a policy for the real robot. But that depends on the system latency on your hardware.

* For `climb_down_policy_path` in distilling the parkour policy.
    Climb-down policy is an add-on to handle climbing down terrain, which was not implemented as the paper submitted. But you can train it by setting `climb_down_prob = 1.0` in `a1_climb_config.py` and train the `a1_climb` task from walking policy.
    We will release an update for climbing down the task for a safer and more error-tolerance gait.

* For data collection (before distillation, after all the specialized skills are done.)
    Please make sure that the `collect.py` is running the `DemonstrationSaver` as the `RunnerCls`. Then the trajectory will be saved in [line 110](https://github.com/ZiwenZhuang/parkour/blob/main/legged_gym/legged_gym/scripts/collect.py#L110). You can modify this as you like.
    You can also run the `collect.py` multiple times or in parallel to get the collection quickly. But typically, 1e6 timesteps is enough.

* About `data_dir` and `scan_dir` field. It is provided to the `rollout_dataset.py` to fetch the pre-collected trajectories. You can use only one attribute.
   For using `data_dir`, you can set it as a list of absolute paths to the `save_dir` which you specified in `collect.py`. In this case, you can start the distillation training without launching a collector. But that does not include the dagger process.
   Using `scan_dir`, it is designed to compute and assign a directory when you launch the dagger trainer. Then, the dagger collector will read this field from the `config.json` file.

* About launching the `collect.py` as a Dagger collector.
    It is correct to set `RunnerCls` to `DaggerSaver`. However due to the code structure, what to specify in the `--load_run` will not be prompted in the dagger trainer's console. You need to find it yourself in the `logs/distill_a1` direction. It is typically the latest run in the directory.

* For visualization, the `--load_run` is the same as mentioned previously.

* About the difference between `distill_{robot}_dagger` and `distill_{robot}`.
    `distill_{robot}_dagger` is used to put those trajectories collected by the dagger collector. You can remove them after the distillation process is done.
    `distill_{robot}` is the directory under `logs` to save all the log dirs.

* About running distillation in one GPU.
    Initially, the code is implemented in one GPU. However, the rendering process in IsaacGym is in serial w.r.t robot. I change the code to multi-process because it is too slow.
    You can run it in one GPU by setting the `runner_class_name` in `a1_field_distill_config.py` back to `OnPolicyRunner`. 
    Remove the configurations about `pretrain_dataset`.
    Tune parameters such as `teacher_act_prob` and `update_times_scale` in the algorithm class. This specifies the dagger process about when and how to switch completely to student acting.
    This might need more debugging because I cannot guarantee the code is still working.

I hope I have answered all your questions.
Best,

xkhainguyen, 2023-10-24:
Hi

Thank you for the detailed responses.

I have a single RTX3070 8GB. The training was working fine. 

However, I ran the `collect.py` with `DemonstrationSaver` as you said but always encountered this 
```Physics Engine: PhysX
Physics Device: cpu
GPU Pipeline: disabled
Segmentation fault (core dumped)
```
I played with cpu, gpu, num_envs (even 1), rollout_storage_length, etc. without any success.

**Is it just because I do not have enough memory or something else?**

I know this is too much to ask but it would be amazing if you can demonstrate the whole pipeline in a more clear way (step by step details or videos).

Thank you very much!

ZiwenZhuang, 2023-11-14:
Hi,

The minimum GPU I've tried is an Nvidia 3080 with 11GB. The out-of-GPU memory issue could be solved by reducing terrain size, e.g. num_rows or num_cols.

I will keep collecting related questions and add instructions in the wiki for the whole pipeline in the near future.

Best,

closed 2023-11-14 by ZiwenZhuang