ZiwenZhuang/parkour #4: Sudden termination of code execution

chenci107, 2023-10-18:
My code suddenly terminates at a certain moment and displays "segment error (core has been dumped)", and the interruption time is also quite random. What is the problem? How should we solve it? thanks
[image]

ZiwenZhuang, 2023-10-18:
Hi, did you check your memory usage? It is likely that the program reaches the limit of the GPU memory or the main memory.

chenci107, 2023-10-18:
Thank you for your reply. My memory capacity is 32G, the GPU used is 3080ti, and the number of parallel environments opened is 2048. I am going to use the 3090 GPU to test.

ZiwenZhuang, 2023-10-18:
No problem! Typically 3080 (10G) for training 2048 robots in parallel is not enough.

closed 2023-10-18 by ZiwenZhuang