ZiwenZhuang/parkour #6: error 'cannot find -lunitree_legged_sdk' in related project's go1 branch

DaojiePENG, 2023-11-23:
When I try to `catkin build` the related project-`unitree_ros_real-go1 branch` from Tsinghua-MARS-Lab, it rises an error 'cannot find -lunitree_legged_sdk'
```
...
Starting  >>> unitree_legged_real                                              
_______________________________________________________________________________
Errors     << unitree_legged_real:make /home/pdj/parkour_ros_real/logs/unitree_legged_real/build.make.001.log
/usr/bin/ld: cannot find -lunitree_legged_sdk
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/unitree_ros.dir/build.make:107: /home/pdj/parkour_ros_real/devel/.private/unitree_legged_real/lib/libunitree_ros.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:1330: CMakeFiles/unitree_ros.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
cd /home/pdj/parkour_ros_real/build/unitree_legged_real; catkin build --get-env unitree_legged_real | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

...............................................................................
Failed     << unitree_legged_real:make           [ Exited with code 2 ]        
Failed    <<< unitree_legged_real                [ 0.3 seconds ]    
...           
```
I have coupied the unitree_legged_sdk to the `src` directory. How can I solve this problem? Thanks!

ZiwenZhuang, 2023-11-23:
Hi,

Did you compile the SDK on Nvidia Jetson? or any onboard machine? I did experience some issues on my x86 laptop. But everything goes well on the Jetson board.

Due to the library changes of Unitree SDK, I cannot create a shared library based on `ros_udp_node.cpp` on the x86 machine.

It would be better to compile this ROS package directly on Nvidia Jetson.

For more detail information, you can check line 41 and line 56 on `unitree_ros_real/unitree_legged_real/CMakeLists.txt`

Also, I used `catkin_make` on ROS1. Maybe ROS version could also lead to problems.

Best,

DaojiePENG, 2023-12-02:
I tried `catkin build` on the onboard jetson nano and the problem 'cannot find -lunitree_legged_sdk' still occured. Then I tried the `catkin_make`, it worked well exccept some warnings. It seems like the ROS version problem. 

Thanks again!

closed 2023-12-02 by DaojiePENG