ZiwenZhuang/parkour #7: The reward of the trained policy `a1_field` is always 0; play error `KeyError: 'climb'`

DaojiePENG, 2023-11-23:
The reward of the trained policy `a1_field` is always 0 from the begining to the end.
```
...
################################################################################
                     Learning iteration 9999/10000                      

                       Computation: 16523 steps/s (collection: 1.199s, learning 0.288s)
               Value function loss: 0.0071
                    Surrogate loss: -0.0000
             Mean action noise std: 706.21
                       Mean reward: 0.00
               Mean episode length: 16.04
            Mean episode rew_alive: 0.0311
      Mean episode rew_frame_alive: 0.0400
Mean episode rew_exceed_dof_pos_limits: -0.0185
Mean episode rew_frame_exceed_dof_pos_limits: -0.0238
Mean episode rew_exceed_torque_limits_i: -0.0031
Mean episode rew_frame_exceed_torque_limits_i: -0.0040
Mean episode rew_legs_energy_substeps: -0.2344
Mean episode rew_frame_legs_energy_substeps: -0.3077
 Mean episode rew_tracking_ang_vel: 0.0002
Mean episode rew_frame_tracking_ang_vel: 0.0002
 Mean episode rew_world_vel_l2norm: -0.0108
Mean episode rew_frame_world_vel_l2norm: -0.0138
        Mean episode timeout_ratio: 0.0000
            Mean episode max_pos_x: 0.7908
            Mean episode min_pos_x: -0.1488
            Mean episode max_pos_y: 0.6007
            Mean episode min_pos_y: -0.5227
    Mean episode n_obstacle_passed: 0.0000
                Mean episode pos_x: 0.3871
Mean episode max_power_throughout_episode: 7562.7012
          Mean episode max_torques: 51.0595
Mean episode torque_exceed_count_substeps_per_envstep: 3.3441
Mean episode torque_exceed_count_envstep: 15.5487
Mean episode out_of_dof_pos_limit_count_substep: 30.7613
--------------------------------------------------------------------------------
                   Total timesteps: 245760000
                    Iteration time: 1.49s
                        Total time: 15273.20s
                               ETA: 1.5s

```
The related command is
`.../parkour/legged_gym$ python legged_gym/scripts/train.py --headless --task a1_field
`
Is there anything I missed? 

---

Despite the 0 reward, then I tried to play the result, it gave a `KeyError: 'climb'` error as shown below:
```
.../parkour/legged_gym$ python legged_gym/scripts/play.py --task a1_field --load_run Nov17_17-20-05_WalkingBase_pEnergySubsteps2e-5_aScale0.5
Importing module 'gym_38' (/home/pdj/parkour_env/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/pdj/parkour_env/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
PyTorch version 1.10.0+cu113
Device count 1
/home/pdj/parkour_env/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/pdj/.cache/torch_extensions/py38_cu113 as PyTorch extensions root...
Emitting ninja build file /home/pdj/.cache/torch_extensions/py38_cu113/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
Setting seed: 1
Using LeggedRobotField.__init__, num_obs and num_privileged_obs will be computed instead of assigned.
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: enabled
Traceback (most recent call last):
  File "legged_gym/scripts/play.py", line 340, in <module>
    play(args)
  File "/home/pdj/anaconda3/envs/parkour_env38/lib/python3.8/site-packages/torch/autograd/grad_mode.py", line 28, in decorate_context
    return func(*args, **kwargs)
  File "legged_gym/scripts/play.py", line 164, in play
    env, _ = task_registry.make_env(name=args.task, args=args, env_cfg=env_cfg)
  File "/home/pdj/parkour_env/parkour/legged_gym/legged_gym/utils/task_registry.py", line 99, in make_env
    env = task_class(   cfg=env_cfg,
  File "/home/pdj/parkour_env/parkour/legged_gym/legged_gym/envs/base/legged_robot_field.py", line 20, in __init__
    super().__init__(cfg, sim_params, physics_engine, sim_device, headless)
  File "/home/pdj/parkour_env/parkour/legged_gym/legged_gym/envs/base/legged_robot.py", line 72, in __init__
    super().__init__(self.cfg, sim_params, physics_engine, sim_device, headless)
  File "/home/pdj/parkour_env/parkour/legged_gym/legged_gym/envs/base/base_task.py", line 84, in __init__
    self.create_sim()
  File "/home/pdj/parkour_env/parkour/legged_gym/legged_gym/envs/base/legged_robot.py", line 238, in create_sim
    self._create_terrain()
  File "/home/pdj/parkour_env/parkour/legged_gym/legged_gym/envs/base/legged_robot_field.py", line 632, in _create_terrain
    self.terrain.add_terrain_to_sim(self.gym, self.sim, self.device)
  File "/home/pdj/parkour_env/parkour/legged_gym/legged_gym/utils/terrain/barrier_track.py", line 614, in add_terrain_to_sim
    starting_height_px = self.add_track_to_sim(
  File "/home/pdj/parkour_env/parkour/legged_gym/legged_gym/utils/terrain/barrier_track.py", line 555, in add_track_to_sim
    obstacle_id = self.track_options_id_dict[obstacle_name]
KeyError: 'climb'

```
What should I do to solve these problems?

zyayoung, 2023-11-23:
> The reward of the trained policy a1_field is always 0

I have the same issue here.

ZiwenZhuang, 2023-11-23:
Hi

> KeyError: 'climb'

We updated all the names from "climb" to "jump" to align with my private implementation. Could it be your old training runs that have the log saying "climb"?

I run the `play.py` script, which does not have such an issue.

> The reward of the trained policy a1_field is always 0.

The reward of each timestep is at the scale of 8e-3, where there are not enough digits to display.

You may update the logging code to show more digits in the terminal or view the training curve using tensorboard. (even during training)

If the training is still too hard, you may try adding `only_positive_reward = False` in `A1FieldCfg.rewards`.

Best,

DaojiePENG, 2023-11-24:
Thanks a lot! After clearing the old `logs` ffile and runing again, it is able to `play`. While the `0 reward problem` still happen envn when I have changed the showing digits to `.6f`. And the playing result seems to be unnormal. The robot keeps reinitilizing and falling in the same pose as shown below with a repeat interval about 0.5s:
[image]
The corresponding console gives messages as follows:
```
...
Loading model from: /home/pdj/parkour_env/parkour/legged_gym/logs/field_a1/Nov24_00-52-36_WalkForward_aScale0.5/model_10000.pt
motor_strength: [1.0780518054962158, 0.9054903984069824, 1.08062744140625, 1.0077260732650757, 1.0462442636489868, 1.0809357166290283, 0.9673910140991211, 0.9069483280181885, 1.0266804695129395, 0.9040271639823914, 1.0861496925354004, 1.0121430158615112]
torque_limits: tensor([25., 25., 25., 25., 25., 25., 25., 25., 25., 25., 25., 25.],
       device='cuda:0')
frame_rate: 2085.5078228232637 command_x: tensor(1.2000, device='cuda:0')
env dones, not because has timeout
{'episode': {'rew_alive': tensor(0.0360, device='cuda:0'), 'rew_frame_alive': tensor(0.0400, device='cuda:0'), 'rew_exceed_dof_pos_limits': tensor(-0.0215, device='cuda:0'), 'rew_frame_exceed_dof_pos_limits': tensor(-0.0239, device='cuda:0'), 'rew_exceed_torque_limits_i': tensor(-0.0010, device='cuda:0'), 'rew_frame_exceed_torque_limits_i': tensor(-0.0011, device='cuda:0'), 'rew_legs_energy_substeps': tensor(-0.0313, device='cuda:0'), 'rew_frame_legs_energy_substeps': tensor(-0.0348, device='cuda:0'), 'rew_tracking_ang_vel': tensor(0.0005, device='cuda:0'), 'rew_frame_tracking_ang_vel': tensor(0.0006, device='cuda:0'), 'rew_world_vel_l2norm': tensor(-0.0186, device='cuda:0'), 'rew_frame_world_vel_l2norm': tensor(-0.0207, device='cuda:0'), 'timeout_ratio': tensor(0., device='cuda:0'), 'max_pos_x': 0.0, 'min_pos_x': 0.0, 'max_pos_y': 0.0, 'min_pos_y': 0.0, 'n_obstacle_passed': tensor(0.), 'pos_x': tensor([0.6968], device='cuda:0'), 'max_power_throughout_episode': 2922.377197265625, 'max_torques': tensor([37.3677], device='cuda:0'), 'torque_exceed_count_substeps_per_envstep': tensor([0.5000], device='cuda:0'), 'torque_exceed_count_envstep': tensor([5], device='cuda:0', dtype=torch.int32), 'out_of_dof_pos_limit_count_substep': tensor([23.4444], device='cuda:0')}, 'time_outs': tensor([False], device='cuda:0')}
env dones, not because has timeout
{'episode': {'rew_alive': tensor(0.0320, device='cuda:0'), 'rew_frame_alive': tensor(0.0400, device='cuda:0'), 'rew_exceed_dof_pos_limits': tensor(-0.0191, device='cuda:0'), 'rew_frame_exceed_dof_pos_limits': tensor(-0.0239, device='cuda:0'), 'rew_exceed_torque_limits_i': tensor(-0.0010, device='cuda:0'), 'rew_frame_exceed_torque_limits_i': tensor(-0.0012, device='cuda:0'), 'rew_legs_energy_substeps': tensor(-0.0103, device='cuda:0'), 'rew_frame_legs_energy_substeps': tensor(-0.0128, device='cuda:0'), 'rew_tracking_ang_vel': tensor(0.0003, device='cuda:0'), 'rew_frame_tracking_ang_vel': tensor(0.0004, device='cuda:0'), 'rew_world_vel_l2norm': tensor(-0.0113, device='cuda:0'), 'rew_frame_world_vel_l2norm': tensor(-0.0141, device='cuda:0'), 'timeout_ratio': tensor(0., device='cuda:0'), 'max_pos_x': 0.0, 'min_pos_x': 0.0, 'max_pos_y': 0.0, 'min_pos_y': 0.0, 'n_obstacle_passed': tensor(0.), 'pos_x': tensor([0.8048], device='cuda:0'), 'max_power_throughout_episode': 1671.83056640625, 'max_torques': tensor([26.8694], device='cuda:0'), 'torque_exceed_count_substeps_per_envstep': tensor([0.5000], device='cuda:0'), 'torque_exceed_count_envstep': tensor([5], device='cuda:0', dtype=torch.int32), 'out_of_dof_pos_limit_count_substep': tensor([24.6875], device='cuda:0')}, 'time_outs': tensor([False], device='cuda:0')}
env dones, not because has timeout
{'episode': {'rew_alive': tensor(0.0280, device='cuda:0'), 'rew_frame_alive': tensor(0.0400, device='cuda:0'), 'rew_exceed_dof_pos_limits': tensor(-0.0167, device='cuda:0'), 'rew_frame_exceed_dof_pos_limits': tensor(-0.0239, device='cuda:0'), 'rew_exceed_torque_limits_i': tensor(-0.0010, device='cuda:0'), 'rew_frame_exceed_torque_limits_i': tensor(-0.0014, device='cuda:0'), 'rew_legs_energy_substeps': tensor(-0.0231, device='cuda:0'), 'rew_frame_legs_energy_substeps': tensor(-0.0329, device='cuda:0'), 'rew_tracking_ang_vel': tensor(0.0003, device='cuda:0'), 'rew_frame_tracking_ang_vel': tensor(0.0004, device='cuda:0'), 'rew_world_vel_l2norm': tensor(-0.0104, device='cuda:0'), 'rew_frame_world_vel_l2norm': tensor(-0.0149, device='cuda:0'), 'timeout_ratio': tensor(0., device='cuda:0'), 'max_pos_x': 0.0, 'min_pos_x': 0.0, 'max_pos_y': 0.0, 'min_pos_y': 0.0, 'n_obstacle_passed': tensor(0.), 'pos_x': tensor([0.7606], device='cuda:0'), 'max_power_throughout_episode': 2310.9130859375, 'max_torques': tensor([27.9775], device='cuda:0'), 'torque_exceed_count_substeps_per_envstep': tensor([0.5714], device='cuda:0'), 'torque_exceed_count_envstep': tensor([5], device='cuda:0', dtype=torch.int32), 'out_of_dof_pos_limit_count_substep': tensor([24.6429], device='cuda:0')}, 'time_outs': tensor([False], device='cuda:0')}
env dones, not because has timeout
...
```
I am not able to test the keyboard functions up to now. 

---

I am trying the `only_positive_reward = False` method and will feedback later.

ZiwenZhuang, 2023-11-24:
Hi,

It's great to hear that.

The problem of frequent resetting because of the termination condition is frequently reached. You may change the code in `play.py` and disable the termination checking during play.

Best,

DaojiePENG, 2023-11-24:
With a insight into the `mean reward` , I found the previous run did get a 0 reward in the end.  
[image]

---

I haven't figure out the resaon why the reward collapse to 0. 

Despite that, I am glad to tell you that, after implementing the `only_positive_reward = False` , I got a final reward of about 10.  And it plays well, showing that the dog runs onto the first stage and then resets. 

Thanks a lot !

closed 2023-11-26 by ZiwenZhuang