ZiwenZhuang/parkour #11: ValueError: operands could not be broadcast together with shapes (593,464,2) (592,464,2)

DaojiePENG, 2023-11-28:
After running the  `a1_jump-virtual` train, I tried to play the result. However, I got a error saying that `operands could not be broadcast together with shapes (593,464,2) (592,464,2) `. The details are as follows:
```
.../parkour/legged_gym$ python legged_gym/scripts/play.py --task a1_jump --load_run Nov26_17-29-57_Skills_jump_virtual_fromNov24_09-41-36
Importing module 'gym_38' (/home/pdj/parkour_env/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/pdj/parkour_env/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
PyTorch version 1.10.0+cu113
Device count 1
/home/pdj/parkour_env/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/pdj/.cache/torch_extensions/py38_cu113 as PyTorch extensions root...
Emitting ninja build file /home/pdj/.cache/torch_extensions/py38_cu113/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
Setting seed: 1
Using LeggedRobotField.__init__, num_obs and num_privileged_obs will be computed instead of assigned.
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: enabled
Traceback (most recent call last):
  File "legged_gym/scripts/play.py", line 341, in <module>
    play(args)
  File "/home/pdj/anaconda3/envs/parkour_env38/lib/python3.8/site-packages/torch/autograd/grad_mode.py", line 28, in decorate_context
    return func(*args, **kwargs)
  File "legged_gym/scripts/play.py", line 165, in play
    env, _ = task_registry.make_env(name=args.task, args=args, env_cfg=env_cfg)
  File "/home/pdj/parkour_env/parkour/legged_gym/legged_gym/utils/task_registry.py", line 116, in make_env
    env = task_class(   cfg=env_cfg,
  File "/home/pdj/parkour_env/parkour/legged_gym/legged_gym/envs/base/legged_robot_field.py", line 20, in __init__
    super().__init__(cfg, sim_params, physics_engine, sim_device, headless)
  File "/home/pdj/parkour_env/parkour/legged_gym/legged_gym/envs/base/legged_robot.py", line 72, in __init__
    super().__init__(self.cfg, sim_params, physics_engine, sim_device, headless)
  File "/home/pdj/parkour_env/parkour/legged_gym/legged_gym/envs/base/base_task.py", line 84, in __init__
    self.create_sim()
  File "/home/pdj/parkour_env/parkour/legged_gym/legged_gym/envs/base/legged_robot.py", line 265, in create_sim
    self._create_terrain()
  File "/home/pdj/parkour_env/parkour/legged_gym/legged_gym/envs/base/legged_robot_field.py", line 651, in _create_terrain
    self.terrain.add_terrain_to_sim(self.gym, self.sim, self.device)
  File "/home/pdj/parkour_env/parkour/legged_gym/legged_gym/utils/terrain/barrier_track.py", line 600, in add_terrain_to_sim
    self.build_heightfield_raw()
  File "/home/pdj/parkour_env/parkour/legged_gym/legged_gym/utils/terrain/barrier_track.py", line 444, in build_heightfield_raw
    heightfield_noise = TerrainPerlin.generate_fractal_noise_2d(
  File "/home/pdj/parkour_env/parkour/legged_gym/legged_gym/utils/terrain/perlin.py", line 68, in generate_fractal_noise_2d
    noise += amplitude * TerrainPerlin.generate_perlin_noise_2d((xSamples, ySamples), (xScale, yScale)) * zScale
  File "/home/pdj/parkour_env/parkour/legged_gym/legged_gym/utils/terrain/perlin.py", line 49, in generate_perlin_noise_2d
    n00 = np.sum(grid * g00, 2)
ValueError: operands could not be broadcast together with shapes (593,464,2) (592,464,2) 
```

---

I also tried other skills like `leap, crawl, tilt`.  The same problem happened with `crawl` while `leap` and `tilt` went well. 

What should I do to fix it? Thanks!

ZiwenZhuang, 2023-11-28:
Hi,

The problem happens because of the track size.

For a simple solution, you may try different numbers of `n_obstacles_per_track` (e.g. 3). Or forcing the `track_block_length` to 2.

Best,

DaojiePENG, 2023-11-28:
Thanks a lot, it works!

closed 2023-11-28 by DaojiePENG