ZiwenZhuang/parkour #12: Policy mean reward of `go1_feild` is always 0 or negtive.

DaojiePENG, 2023-11-29:
I trained the `go1_feild` script directly and got 0 mean reward from beginning to the end. This happen when training the `a1_feild` script. Then follow the experience of `a1_feild`, I change `only_positive_rewards = True` to `only_positive_rewards = Flase`.  However, the mean reward raise only to aound -1.6 in the end. The image below shows the situation.

 [image]

Then I played the policy. The go1 in `0 reward` case shows a method of stuck in strange pose and almost without any motion. The go1 in `-1.6 reward` case keeps hitting the ground and resetting. The image below shows the `-1.6 reward` case.

[image]

I wonder why this happens? What shall I do to overcome this? Thanks!

DaojiePENG, 2023-12-01:
I also tried comment  `latency_range` and `mu_activation = None` to make it more like the `a1` case. But it still didn't work, always have negtive mean reward and can't be palyed normally. Actually, my changes make the result worse.
[image]

ZiwenZhuang, 2023-12-01:
Hi,

The config files for Go1 currently are examples of the oracle policies for the CoRL live demo. It shows how the checkpoints for the demo are trained but might be too strict to train from scratch.

Please refer to the commit before Nov. 08, where the config should be easier to train from scratch.

Best,

DaojiePENG, 2023-12-01:
Oh, I see. Thanks a lot!