ZiwenZhuang/parkour #13: Error：Position Protection

XiaoWZENG, 2023-11-30:
Hello, have you ever meet the problem with position protection of the leg 2 calf when run the node "roslaunch unitree_legged_real robot.launch" in the terminal. At the same time I also can hear the vibration sounds from all the joint motors.
Thank you very much!

ZiwenZhuang, 2023-11-30:
Hi,

Are you running on Unitree A1? The position protect error is to protect the joints from hitting the mechanical limit. You can disable it by setting the `position_protect_limit` parameter to zero.

**But NOTICE that this could potentially cause serious hardware damage.**

I'm not sure what kind of vibration sound it is, but the mechanical build of Unitree A1 is not as good as Unitree Go1, which produces vibration noise under high speed and non-smooth action output.

For a better solution, you may need to finetune your policy and make the action output smoother.

Also, we experienced quite a few motor-reset problems during real-robot testing and caused serious motor damages. This is one of the reasons why we won't release the demo checkpoint for Unitree A1.

Best,

XiaoWZENG, 2023-12-01:
Thank you for your kindly reply. I am running the demo on Unitree Go1 robot, and use the well trained policy in the "go1_ckpts " file of your repository. The vibration noise sounds like the vibration of a mobile phone.
I am not sure whether it is caused by the mistake operation at the initial state, I tried both damping mode and joint level control mode, by pressing the bottom L2+B / L2+A+START.

Best,

ZiwenZhuang, 2023-12-01:
Hi,

It could be the inconsistency between the sensory latency from running on NX and running on your laptop.

How do you connect your laptop to the system? Using WiFi is not recommended, since WiFi might result in non-stable data transmission.

To verify this, you may try playing the parkour policy in the simulator while overriding the proprioception latency.

Best,

XiaoWZENG, 2023-12-01:
Amazing, it works! I use a Ethernet cable to connect my PC to the Go1 robot.
Thanks!
Best,