ZiwenZhuang/parkour #16: ‘Illegal instruction (core dumped)’ problem when running `go1_visual_embedding.py` or `a1_ros_run.py`

DaojiePENG, 2023-12-04:
When I tried to run the command in `T_visual` and `T_gru`, they raised `Illegal instruction (core dumped)` and ended. 

```
(parkour_venv) unitree@nx:~/parkour_deploy$ python go1_visual_embedding.py --logdir Nov02_16-18-16_674k_distill_crawljumpjumpleap_vDelay0.25-0.30_camPitch0.52_depthMax2.0_jumpNov02_11-07-49_leapVel1.5_jumpOffset0.05_oracleResetWhenSkill_fricMax2.0_noTanh_fromOct28_16-16-58
Illegal instruction (core dumped)
(parkour_venv) unitree@nx:~/parkour_deploy$ python a1_ros_run.py --mode upboard --logdir Nov02_16-18-16_674k_distill_crawljumpjumpleap_vDelay0.25-0.30_camPitch0.52_depthMax2.0_jumpNov02_11-07-49_leapVel1.5_jumpOffset0.05_oracleResetWhenSkill_fricMax2.0_noTanh_fromOct28_16-16-58
Illegal instruction (core dumped)
(parkour_venv) unitree@nx:~/parkour_deploy$ 
```

---

The `unitree_legged_real` command went well. I am presently running in the `unitree@nx`. I also tried with `unitree@unitree-desktop`, the same problem occured. 

What's the problem and how to deal with it? Thanks!

DaojiePENG, 2023-12-09:
I thought it was because I didn't go into the `develop` mode, so the core might be occupied by unitree's process. Then I tried `L2+A+A, L2+B` , `L1+L2+start` to go into the normal mode and tried the command again. However, the issue of `T_visual and T_gru` was not solved.

ZiwenZhuang, 2023-12-09:
Hmmm, I haven't experienced this before. It seems like some version problems when running the Python program.

Could you locate which function call that leads to the error? e.g. creating a pytorch tensor, tensor to cuda, etc.

DaojiePENG, 2023-12-09:
I am not able to locate the `core dump` source code. 

Good news is that I tried to create a new venv and the problem was solved. I finally realsed that the reason might be the wrong `numpy` version. I removed the version requirement at the first time when I make the venv. 
Now it's able to run and render the corresponding information normally. 

Thanks a lot for your inspiration!

closed 2023-12-09 by DaojiePENG