ZiwenZhuang/parkour #19: Sim2real transfer, udp_recv error

Anchoret13, 2023-12-12:
After first build, I was able to run `roslaunch unitree_legged_real robot.launch`, but after restart, I keep getting `udp_recv error: -1 message translate might lost once` and unable to get the info from robot any more. 

Rebuild didn't solve the problem. The sdk version is 3.8.6, and I'm pretty sure all packages were built successfully.

ZiwenZhuang, 2023-12-21:
Please check whether you used the correct SDK and ROS package for the correct robot model (A1 or Go1)
 
Then, please refer to Unitree's official documentation and check whether you set the robot to the correct mode.

Anchoret13, 2024-01-04:
Hardware issue, problem solved

closed 2024-01-04 by Anchoret13

DecaYale, 2024-04-03:
> Hardware issue, problem solved

How do you solve this? Thx.