ZiwenZhuang/parkour #20: Unable to stand up from crouching, requiring legs to be stretched

DaojiePENG, 2023-12-13:
I found the go1 was unable to stand up directly from crouching state(L1+L2+Start, this command closes unitree's controller into normal mode) with the provided policy. 
The `T_ros, T_visul` went well while the `T_gru` gave something like the follows:
```
...
Actor RNN: Memory(
  (rnn): GRU(176, 256)
)
Critic RNN: Memory(
  (rnn): GRU(81, 256)
)
[INFO] [1702284120.606872]: duration: 0.02, motor Kp: [40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0], motor Kd: [0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5]
[INFO] [1702284120.622811]: Using stand only model, please make sure the proprioception is 48 dim.
[INFO] [1702284120.630576]: Using stand only model, -36 to -24 must be joint position.
[INFO] [1702284120.743915]: UnitreeA1Real.low_state_buffer acquired, stop waiting.
[INFO] [1702284120.752136]: Robot standing up, please wait ...
[INFO] [1702284120.881907]: a1_ros_run recieved forward depth embedding.
max joint error (rad): 1.2932434082031252
```

I have to lift and strech out it's legs to make it trying to stand successfully. 

I saw from other issues that I may need to use `(L2+B then L2+A+START)`. However, I failed to use this. It seems that nothing changes when I press `L2+A+START` together and  `L2+A` then `START` starts the Unitree's controller. 

I have tried when the go1 is `L2+A` lockedd. Both low and high stand locking states lead to `T_gru` outputs as follows:

```
...
[ WARN] [1702138382.317291680]: Cmd lost, position control as motor damping. DO NOT use this as a normal operation.
[ERROR] [1702138382.327995616]: udp_recv error: -1, message translate might lost once.
[ERROR] [1702138382.437948096]: udp_recv error: -1, message translate might lost once.
[ERROR] [1702138382.547973920]: udp_recv error: -1, message translate might lost once.
[ERROR] [1702138382.657931168]: udp_recv error: -1, message translate might lost once.
[ERROR] [1702138382.767965280]: udp_recv error: -1, message translate might lost once.
[ERROR] [1702138382.877974912]: udp_recv error: -1, message translate might lost once.
...
```

---

I am not sure what's the problem. It seems like that the policy doesn't provide enough torque for the go1's stand-up evoking.

Thanks for any advices!

ZiwenZhuang, 2023-12-14:
The stand-up procedure is hard-coded. I used a conservative PD factor. You may increase the values so the motor will exert more torque to stand up.

You may check [standup_procedure in a1_ros_run.py](https://github.com/ZiwenZhuang/parkour/blob/main/onboard_codes/a1_ros_run.py#L106) and adjust the parameters accordingly.

For the error `[ERROR] [1702138382.437948096]: udp_recv error: -1, message translate might lost once.` Please check your hardware system and the SDK. This means the ROS node fails to receive the low state from the SDK if you keep seeing this.

DaojiePENG, 2023-12-14:
> The stand-up procedure is hard-coded. I used a conservative PD factor. You may increase the values so the motor will exert more torque to stand up.
>
> You may check [standup_procedure in [...]

I have tried to change the PD factor. It helped the state-maintance after stand-up while it did no help to the stand-up evoking process. 

I notice the `max joint error (rad): 1.2932434082031252` is quite big. Sadly, the go1 seldom tries to evoking to reduce the error. 

The go1 starts from  crouching state as shown below:

[image]

I am not sure if this is a proper pose for running the command. But it is presently the only state I find that is able to evoking(Of course, with my aid). 

I wonder what pose do you use and how to get the go1 to that state?

DaojiePENG, 2023-12-14:
Another problem I meet is that when changed into skill-policy, the robot is easy to be influenced though I have increased the kp here( in a1_ros_run.py):
```
...
elif args.mode == "upboard":
...
    if unitree_real_env.low_state_buffer.wirelessRemote.btn.components.L2 or unitree_real_env.low_state_buffer.wirelessRemote.btn.components.R2:
                    unitree_real_env.publish_legs_cmd(unitree_real_env.default_dof_pos.unsqueeze(0), kp= 100, kd= 0.5)
                    rospy.signal_shutdown("Controller send stop signal, exiting")
...
```

The legs of go1 in skill-policy seems not to have enough eager to keep the standing pose. As a result, almost every stop end up with an odd pose similar like the follows:
[image]

Sometimes it even falls down.

ZiwenZhuang, 2023-12-21:
Where did you change the PD factor? The snippet you referred to is not related to `standup_procedure`.

ZiwenZhuang, 2023-12-21:
Did you launch the ROS node with `dryrun:=false`? To make sure the user is aware of potential motor movement, the `dryrun` argument in the launch file is set to `True` by default and the user has to launch the process by specifying the argument explicitly.

Otherwise, the ROS node will only send all proprioception sensory data in the ROS network but not sending any motor command to the robot.

DaojiePENG, 2023-12-22:
> Did you launch the ROS node with `dryrun:=false`? To make sure the user is aware of potential motor movement, the `dryrun` argument in the launch file is set to `True` by default and the user has to [...]

Yes, I launch it with `dryrun:=false`. 

I found that the legs of go1 need to be slightly streched (just a bit makes the whole difference ) to begin the stand process. This may not be necessary if I use `L2+B` directly from the standing-lock state. I think the init-checking of standing function is a bit strict. If I use "L2+B" when go1 is low, there will be some issues with its operation as the legs will be pressed under the body. I didn't realize this before cause unitree's primary controler can stand under that circumstance.

Thanks a lot for your advice!

closed 2023-12-22 by DaojiePENG