ZiwenZhuang/parkour #21: Walking training always fails.

Wtrwater, 2023-12-21:
Similar to issue 7, the rewards of `a1_field` is always 0 and `play.py` repeatedly reset. Even if I add `only_positive_reward = False` in `A1FieldCfg.rewards`. But with the addition of this code, play.py will not restart again, and the robot dog will always stand in place, rather than walk forward. I'm wondering if I did something wrong?

These are the details:
1. IF I ADD `only_positive_reward = False` in `a1_field_config.py `like:

```
class rewards( A1RoughCfg.rewards ):

        class scales:

            tracking_ang_vel = 0.05

            world_vel_l2norm = -1.

            legs_energy_substeps = -2e-5

            legs_energy = -0.

            alive = 2.

            # penalty for hardware safety

            exceed_dof_pos_limits = -1e-1

            exceed_torque_limits_i = -2e-1

        soft_dof_pos_limit = 0.01

        only_positive_reward = True      # here I add the code 
```
after 10000 iterations, the dog can stand up with close to 100% probability like this:
[image]
[image]

2. IF I DON'T ADD the code, the play.py will repeatedly reset and have similar output in issue 7. I tried to comment the check part in the code like follow, but it doesn't work.
```
        # if dones.any():
        #     agent_model.reset(dones)
        #     print("env dones,{} because has timeout".format("" if env.time_out_buf[dones].any() else " not"))
        #     print(infos)
```

Wtrwater, 2023-12-21:
Or any pt files that have been trained to walk would be appreciated : ). THANKS

Wtrwater, 2023-12-21:
Sometime it performed like this:
[image]

ZiwenZhuang, 2023-12-21:
From my experience, it shouldn't be that difficult to train it walking...
To address some of your questions:
> IF I ADD `only_positive_reward = False` in `a1_field_config.py` like:
You may roll back to use the `a1_field_config.py` file at the initial commit. Maybe try adding `only_positive_reward = False` also.
Also, maybe setting `soft_dof_pos_limit = 0.9` will help making the training easier.

> IF I DON'T ADD the code, the play.py will repeatedly reset and have similar output in issue 7. I tried to comment the check part in the code like follow, but it doesn't work.
As far as I know, the termination conditions have already been overridden by the code in `play.py`. Maybe setting `env_cfg.asset.terminate_after_contact_on = []" will stop the frequent resetting problem.

closed 2023-12-22 by Wtrwater

Wtrwater, 2023-12-23:
Thank you very much for your answer to my question. I also tried to set `soft_dof_pos_limit = 0.9` without modifying `only_positive_reward`. The training was also very successful. The robot learned to walk completely. I'm wondering what does the `soft_dof_pos_limit` mean?

[image]
Fig. Purple: change `soft`, Gray: `soft` + `only`, yellow: no change

Best wishes.

ZiwenZhuang, 2023-12-24:
You may use a global search using your code editor. `soft_dof_pos_limit` scales down the joint limit based on the urdf to prevent resulting the policy that hits the actual joint limit on the real robot.

asibatian, 2024-07-03:
> It works! Thanks!
> 
> I downloaded the first commit and changed it as u said. Finally it got a reward about 28 and successfully walk.
> 
> ![image](https://private-user-images.githubusercontent.com/106796041/292410298-030ae421-a105-452b-a6d5-7a1cbfa10683.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjAwMTU2NDIsIm5iZiI6MTcyMDAxNTM0MiwicGF0aCI6Ii8xMDY3OTYwNDEvMjkyNDEwMjk4LTAzMGFlNDIxLWExMDUtNDUyYi1hNmQ1LTdhMWNiZmExMDY4My5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcwM1QxNDAyMjJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04ZjM2MzAwZmFlYzc5MmRlYzg3NGU2ZDU5Mzc5MjQ5MzczNzgyMjM0ZGUzMGJhOGZhZjUxMmJkY2E4ZTAxODYwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.T9xLXeuuZ0s9PlwjnVi90dk2XXUQg_YyGfr2CRTPyKQ)

I also meets this problem and I tried the method above:
[image]
And using command `ython legged_gym/scripts/train.py --headless --task a1_field --run_name train-03` to train.
Finally, I got reward about 35:
[image]

But when I play the result, a1 still repeats termination when the robot hit the wall.
So I tried:

> Maybe setting `env_cfg.asset.terminate_after_contact_on = []` between line 123 and line 124 will stop the frequent resetting problem.
[image]

Then I play using `python legged_gym/scripts/play.py --task a1_field --load_run train-03`, a1 does not reset again and again, but it stops at the step:
[image]
[image]

I am not sure what is wrong. Do I miss something?
Hope to get your suggestion @Wtrwater @MarkFzp @ZiwenZhuang , thank you!

asibatian, 2024-11-05:
> @asibatian I encountered the same problem as you, did you finally solve it?

Sorry, I did not solve this problem. I do not have two GPU so that I give up to re-implement this work. 
This project is good and I hope you well.