ZiwenZhuang/parkour #22: About RND and Oracles w/o Soft Dyn

yingchengyang, 2023-12-27:
Thanks a lot for providing such wonderful work and clean codes! I'm curious about the results of RND and Oracles w/o Soft Dyn in Fig.7 and Fig. 9 in the paper (https://robot-parkour.github.io/resources/Robot_Parkour_Learning.pdf). In my opinion, they both directly train in the hard dynamics constraints by PPO and PPO+RND respectively, is that right? Also, Where is the code about RND in this repo?

Thanks again!

ZiwenZhuang, 2023-12-28:
Yes, both PPO and PPO+RND are trained directly using hard dynamics constraints.

Due to the code structure, the implementation of RND is quite messy in our private repo. Thus, we remove the RND implementation in the public repository.

yingchengyang, 2023-12-29:
Thanks a lot!

closed 2023-12-29 by yingchengyang