ZiwenZhuang/parkour #24: About the success rate reported in the paper

yingchengyang, 2024-01-06:
Thanks a lot for providing such wonderful work and clean codes! I'm curious about the results of the success rate reported in the paper. Are these success rates evaluated in the real world? Or can we evaluate the success rate in the simulator when training a single skill (like jump, or leap)? It seems that the running code only reports the defined rewards.

Thanks again!

ZiwenZhuang, 2024-01-14:
In simulation, we test the success rate by playing the policy in a batch of 10 robots and let each of them run 10 times.

In real-world, we test the success rate by running the policy on the robot for about 10 times and count the number of successes. During climbing(jumping) task, we allow the robot to retry for at most 3 times as long as it does not fall down.

yingchengyang, 2024-01-14:
Thanks a lot! In the simulation, how can we test the success rate? Do we test the trained agent in the hard constraint environment?