ZiwenZhuang/parkour #27: About training Go1

yingchengyang, 2024-01-21:
Thanks a lot for providing such wonderful work and clean codes! I'm curious about the results of Go1.

For running A1 in the simulation via PPO, almost all tasks (field, crawl, jump, leap, tilt) can get rewards of ~20. However, when I run PPO for training Go1, the training is not ideal, some tasks get always 0 and some tasks get negative returns. 

Are there any specific parameters that need to be set for Go1? What is the expected cumulative reward for Go1 tasks?

Thanks again!

ZiwenZhuang, 2024-01-21:
We didn't train the policy for Go1 from scratch.

In fact, most policies trained on A1 can be used on Go1 model directly. Thus, we prepared our CoRL demo by finetuning the A1 expert policies on Go1 model and training the onboard parkour policy with new camera configuration.

yingchengyang, 2024-01-21:
Thanks a lot!

closed 2024-01-21 by yingchengyang