ZiwenZhuang/parkour #29: How to check the position of front camera ?

hakieh, 2024-01-27:
hi,
I am trying to train the policy with go1, but I do not know how to check whether the position of the front camera is in correct position. In other words, after setting a  position in the config file, how can I know the position of the front camera in the Isaac gym environment?

hakieh, 2024-01-27:
I mean that it seems that I can't see the camera in the simulated environment

ZiwenZhuang, 2024-01-27:
There is a debug option in the config file. By setting `env_cfg.viewer.draw_debug_vis = True` and making sure `forward_depth` is in one of your `obs_components` during `play.py`, you will see a RGB axis near the robot in the visualization.

For the visualization implementation, please check [line 737](https://github.com/ZiwenZhuang/parkour/blob/main/legged_gym/legged_gym/envs/base/legged_robot_field.py#L737) `_draw_sensor_vis` in `legged_robot_field.py`.

hakieh, 2024-02-07:
thanks a lot

closed 2024-02-07 by hakieh