ZiwenZhuang/parkour #30: Deploy to Go1 with onboard computation

RolandZhu, 2024-01-30:
Thank you for open-sourcing the code; it's impressive work! I'm working on deploying the network to Go1 EDU using your ros_real implementation. Could you provide details on setting up ROS and other dependencies on the Jetson platform? Are there any specific instructions or documentation for this process?

ZiwenZhuang, 2024-01-31:
Please refer to [Deploy.md](https://github.com/ZiwenZhuang/parkour/blob/main/Deploy.md), where the link to the ros package is mentioned.

hakieh, 2024-02-28:
Hello, I am also trying to deploy the model to Go1 EDU, have you been successful before? I followed the steps of the deploy.md file, but when it started, Go1 couldn't stand up and only wobbled.I don't know what the problem is.

ZiwenZhuang, 2024-02-28:
Hi, the stand-up procedure is a fixed function in the onboard script. Please check the `standup_procedure` method in [a1_ros_run.py](https://github.com/ZiwenZhuang/parkour/blob/main/onboard_codes/a1_ros_run.py#L106). If the legs are not strong enough, simply increase the `kp` or the `angle_tolerance` in [line 288 of a1_ros_run.py](https://github.com/ZiwenZhuang/parkour/blob/main/onboard_codes/a1_ros_run.py#L288)

hakieh, 2024-02-29:
> Hi, the stand-up procedure is a fixed function in the onboard script. Please check the `standup_procedure` method in [...]

Thanks for your reply! I have tried the method you said, but still can't stand up, and I noticed a problem. In the T_ros terminal, the following information will be constantly refreshed. Is it a problem that it cannot stand up? After I turn on the power supply of the go1 robot dog, the sequence of keys is [L1 + A] [L1 + A] [L1 + B] [L1 + L2 + start] to make the dog lie down.
[image]

hakieh, 2024-02-29:
> > Hi, the stand-up procedure is a fixed function in the onboard script. Please check the `standup_procedure` method in [a1_ros_run.py](https://github.com/ZiwenZhuang/parkour/blob/main/onboard_codes/a1_ros_run.py#L106). If the legs are not strong enough, simply increase the `kp` or the `angle_tolerance` in [line 288 of a1_ros_run.py](https://github.com/ZiwenZhuang/parkour/blob/main/onboard_codes/a1_ros_run.py#L288)
> 
> Thanks for your reply! I have tried the method you said, but still can't stand up, and I noticed a problem. In the T_ros terminal, the following information will be constantly refreshed. Is it a problem that it cannot stand up? After I turn on the power supply of the go1 robot dog, the sequence of keys is [L1 + A] [L1 + A] [L1 + B] [L1 + L2 + start] to make the dog lie down. ![12](https://private-user-images.githubusercontent.com/82552207/308804835-188a3d7c-9913-458e-b694-3b384cab09d1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkxODkyODMsIm5iZiI6MTcwOTE4ODk4MywicGF0aCI6Ii84MjU1MjIwNy8zMDg4MDQ4MzUtMTg4YTNkN2MtOTkxMy00NThlLWI2OTQtM2IzODRjYWIwOWQxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjI5VDA2NDMwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTUyMTBjMmRmZDJlMjc0ZmM4NWU2Mjg5ZjZlM2ZjY2Q1NTAxZjhjMDNlNDg5Nzg1ODVjZDYzNTk3ZmNjNWFkZmMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ukTpSGk9uNbl61CreEaOwVXk_HKOFXyUvv5ThjvUkX8)

Thank you! I've already solved the problem

closed 2024-03-14 by ZiwenZhuang

DecaYale, 2024-04-01:
How do you solve this problem? Is this caused by the insufficient computation of nano? I just found out they need a NX for go1.