ZiwenZhuang/parkour #32: No such file or directory: Your trained walk model directory

fanyahao1, 2024-03-17:
Hello, when I use command line `python legged_gym/legged_gym/scripts/train.py --task a1_down --headless` to run a1_down or a1_jump, etc, there is a problem as follows:
```
Traceback (most recent call last):
  File "legged_gym/legged_gym/scripts/train.py", line 49, in <module>
    train(args)
  File "legged_gym/legged_gym/scripts/train.py", line 44, in train
    ppo_runner, train_cfg = task_registry.make_alg_runner(env=env, name=args.task, args=args, env_cfg=env_cfg)
  File "/root/parkour/legged_gym/legged_gym/utils/task_registry.py", line 171, in make_alg_runner
    resume_path = get_load_path(log_root, load_run=train_cfg.runner.load_run, checkpoint=train_cfg.runner.checkpoint)
  File "/root/parkour/legged_gym/legged_gym/utils/helpers.py", line 132, in get_load_path
    models = [file for file in os.listdir(load_run) if 'model' in file]
FileNotFoundError: [Errno 2] No such file or directory: '/root/parkour/legged_gym/logs/field_a1/{Your trained walk model directory}'
```
And I find the configure in class A1JumpCfgPPO.runner:
```
class runner( A1FieldCfgPPO.runner ):
        policy_class_name = "ActorCriticRecurrent"
        experiment_name = "field_a1"
        resume = True
        load_run = "{Your traind walking model directory}"
        load_run = "{Your virtually trained jump model directory}"
```
But I did not have a trained jump model and I can`t. 

And when I use command line `python legged_gym/legged_gym/scripts/train.py --task a1_field --headless`, it dose works, but in every episode, the Mean Reward is 0. I can not find the reason.

ZiwenZhuang, 2024-03-18:
For the safety issue of A1 hardware, we will not provide the checkpoint and training log for Unitree A1.

For training the walking and parkour skills, please refer to [this issue](https://github.com/ZiwenZhuang/parkour/issues/7). The files provided are examples of how you can launch the training process.

fanyahao1, 2024-03-21:
Okey ,thank you for your reply!

closed 2024-03-21 by fanyahao1

1242713693, 2024-12-20:
> class runner

So you can only run the single skill a1_field in the simulation?