ZiwenZhuang/parkour #37: Where is the a1_climb_config.py?

DecaYale, 2024-04-16:
Where is the a1_climb_config.py? I cannot find it in the current repo.

ZiwenZhuang, 2024-04-17:
`climb` task is renamed as `jump` task. so as the `a1_climb_config.py` is moved to `a1_jump_config.py`.

BTW, where did you find the code requesting `a1_climb_config.py`? I'm just checking if I missed anything.

DecaYale, 2024-04-17:
Thank you for your kind reply. You may need to update the readme part indicated by the following link. https://github.com/ZiwenZhuang/parkour/blob/main/legged_gym/README.md#:~:text=scirpt%20with%20task-,a1_climb,-%2C%20a1_leap%2C

BTW. I found the firmware of go1 may have been updated. We need to press L1+L2+start to enter the correct mode for locomotion developing (not the keys L2+B then L2+A+START you mentioned before).

closed 2024-04-17 by DecaYale