ZiwenZhuang/parkour #39: Question about contact force

shivanshpatel35, 2024-05-19:
Hi,

The code uses `self.contact_forces` to compute the reward. It is initialized [here](https://github.com/ZiwenZhuang/parkour/blob/main/legged_gym/legged_gym/envs/base/legged_robot.py#L586). But I cannot find where it is updated after each simulation step. Can you please point me to the relevant part of the code?

ZiwenZhuang, 2024-05-21:
It is automatically updated by isaacgym while doing simulation, which is the same as `self.all_root_states`.

ZiwenZhuang, 2024-05-26:
By checking the built-in config parsing implementation, the "contact_collection" is already set to `2` by default. I'm afraid that you didn't change anything.

> Actually, I was able to get contact forces by setting setattr(sim_params.physx, "contact_collection", gymapi.ContactCollection(2)).

BTW, where do you put this additional code line?