ZiwenZhuang/parkour #40: Problems with python envs while following the tutorial

songsq21, 2024-05-25:
Hello, I am recently discovering on your robot parkour learning, trying to implement it on my computer.
I am following the `legged_gym/README.md` tutorial and after I followed the instructions on the **Installation** part, 
I tried to implement the `python legged_gym/scripts/train.py --headless --task a1_field` command, found out that
[image]
I solved this by altering the corresponding file at `isaacgym` downloaded, reimplemented the command and found that
[image]
I installed `tensorboard` with pip and found that
[image]
I looked up docs on `disutils` and found out that this might be the issue on its version, but `pip install disutils` suggested that
`ERROR: Could not find a version that satisfies the requirement distutils (from versions: none)`
**I thought this would be some issue on the version of other packages such as `tensorboard`, `numpy` or else, but I have no idea to what extent should I downgrade them in order to solve this `disutils` problem.**
Do you have some ideas about this? Thanks a lot!

ZiwenZhuang, 2024-05-26:
That is the problem with `setuptools` package, please refer to some discussions like [this](https://stackoverflow.com/questions/69919970/no-module-named-distutils-but-distutils-installed)

For short, `pip install setuptools==59.5.0` or `pip install setuptools`

songsq21, 2024-05-26:
Thanks for your help! Now I've solved this problem.

closed 2024-05-26 by songsq21