ZiwenZhuang/parkour #42: Hardware Deployment for Unitree Go1

Sg7400, 2024-06-05:
Thank you very much for sharing your resulting code.

I am hardware testing your results using a Unitree Go1 robot.

I only modified 1 line in your code.
I modified angle_tolerance=0.2 in line 289 of a1_ros_run.py to 2.0 and tested it.

There seem to be some issues. 
Could you tell me where I need to change the code?

[Video-1]

https://github.com/ZiwenZhuang/parkour/assets/83255283/e5143c05-6e83-466a-8b1b-32b2836a0b07

1. After running python a1_ros_run.py --mode upboard --logdir Nov02..., the robot does not wake up immediately and starts shaking.
   And after a few seconds, the robot stands up.
2. And when the robot stands up, its upper body is tilted because it cannot straighten its hind legs.

[Video-2]

https://github.com/ZiwenZhuang/parkour/assets/83255283/bc6fa6d2-88e7-4ff1-8834-68fb5c8dd9bc

3. When I control the robot with the joystick, the robot becomes unstable and falls.

Could you tell me where the problem is?

Thank you.

ZiwenZhuang, 2024-06-05:
Did you turn off the built-in control program? (For go1, some buttons on the joysticks must be pressed before you launch any user program)

Maybe refer to the documentation of go1.

Sg7400, 2024-06-07:
L2+A, L2+A, L2+B, L1+L2+Start Pressed the button on the remote controller and started.

I think there is a problem with UDP communication, so I will look into the cause further.

Thank you.

Indiasuperman, 2024-06-30:
sorry to bother you，I want to ask did you just trian the a1 model or go1 in simulation?

ZiwenZhuang, 2024-07-11:
Yes, I trained model both on A1's urdf and Go1's urdf.

Zhihaibi, 2024-09-30:
> Press the L2+A, L2+A, L2+B, L1+L2+Start button on the remote control and start in low-level control mode.
> 
> I think there is a problem with UDP communication, so I will look into the cause further.
> 
> Thank you.

Hi, I have the same situation as you. Go1 keeps shaking at the start of the procedure. Please let me know if you have solved it. Thanks.

tw0127, 2024-10-06:
> Press the L2+A, L2+A, L2+B, L1+L2+Start button on the remote control and start in low-level control mode.
> 
> I think there is a problem with UDP communication, so I will look into the cause further.
> 
> Thank you.
Hi, I have the same problem with go2. Please let me know if you have solved it. 
Thanks.