ZiwenZhuang/parkour #44: Segmentation Fault when executing collect.py

wang414, 2024-07-06:
Thanks for your excellent work. I am facing segmentation fault while running ``python legged_gym/scripts/collect.py --headless --task a1_distill``.  I traced back for the problem, and found that it was caused by the forward_depth observation when the code is creating for the environment. Do you have any idea about it. 

My GPU is A100-40G, pytorch version is 1.10.0+cu113.

AlorithmKing, 2024-07-11:
i meet the same issue when i run  python legged_gym/scripts/train.py --task a1_leap

Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
Setting seed: 1
Using LeggedRobotField.__init__, num_obs and num_privileged_obs will be computed instead of assigned.
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: enabled
Segmentation fault (core dumped)

ZiwenZhuang, 2024-07-11:
> i meet the same issue when i run python legged_gym/scripts/train.py --task a1_leap
> 
> Building extension module gymtorch... Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N) ninja: no work to do. Loading extension module gymtorch... Setting seed: 1 Using LeggedRobotField.**init**, num_obs and num_privileged_obs will be computed instead of assigned. Not connected to PVD +++ Using GPU PhysX Physics Engine: PhysX Physics Device: cuda:0 GPU Pipeline: enabled Segmentation fault (core dumped) My GPU is 4090 torch==2.0.0+cu118 torchvision==0.15.0+cu118 torchaudio==2.0.0+cu118

It seems you forgot to add `--headless` argument in a headless server?

ZiwenZhuang, 2024-07-11:
> Thanks for your excellent work. I am facing segmentation fault while running `python legged_gym/scripts/collect.py --headless --task a1_distill`. I traced back for the problem, and found that it was [...]

If it is the depth rendering problem, please make sure `*Cfg.sim.no_camera` is set to `False` in your config file.