ZiwenZhuang/parkour #45: installation

tomjur, 2024-07-10:
Hi,

I tried to install on a new conda env by following the installation instructions.

The problem is that when I try to run the example script 
`python legged_gym/scripts/train.py --headless --task a1_field`

I get that the numpy version (1.24.5) is calling deprecated functions within the IsaacGym Preview:
`Traceback (most recent call last):
  File "legged_gym/legged_gym/scripts/train.py", line 36, in <module>
    from legged_gym.envs import *
  File "/home/tomjur/legged_gym/legged_gym/envs/__init__.py", line 33, in <module>
    from .base.legged_robot import LeggedRobot
  File "/home/tomjur/legged_gym/legged_gym/envs/base/legged_robot.py", line 37, in <module>
    from isaacgym.torch_utils import *
  File "/home/tomjur/IsaacGym_Preview_4_Package/isaacgym/python/isaacgym/torch_utils.py", line 135, in <module>
    def get_axis_params(value, axis_idx, x_value=0., dtype=np.float, n_dims=3):
  File "/home/tomjur/miniconda3/envs/parkour/lib/python3.8/site-packages/numpy/__init__.py", line 305, in __getattr__
    raise AttributeError(__former_attrs__[attr])
AttributeError: module 'numpy' has no attribute 'float'.
`np.float` was a deprecated alias for the builtin `float`. To avoid this error in existing code, use `float` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.
The aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:
    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations`

Since the code is within Isaac, I can only solve the issue by changing the numpy version to pre-deprecation (numpy 1.19), but this causes the following error:

`Traceback (most recent call last):
  File "legged_gym/legged_gym/scripts/train.py", line 36, in <module>
    from legged_gym.envs import *
  File "/home/tomjur/legged_gym/legged_gym/envs/__init__.py", line 33, in <module>
    from .base.legged_robot import LeggedRobot
  File "/home/tomjur/legged_gym/legged_gym/envs/base/legged_robot.py", line 38, in <module>
    from isaacgym import gymtorch, gymapi, gymutil
  File "/home/tomjur/IsaacGym_Preview_4_Package/isaacgym/python/isaacgym/gymtorch.py", line 109, in <module>
    _import_gymtorch()
  File "/home/tomjur/IsaacGym_Preview_4_Package/isaacgym/python/isaacgym/gymtorch.py", line 45, in _import_gymtorch
    gt = torch.utils.cpp_extension.load(name="gymtorch", sources=sources, extra_cflags=cflags, verbose=True)
  File "/home/tomjur/miniconda3/envs/parkour/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 1124, in load
    return _jit_compile(
  File "/home/tomjur/miniconda3/envs/parkour/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 1337, in _jit_compile
    _write_ninja_file_and_build_library(
  File "/home/tomjur/miniconda3/envs/parkour/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 1423, in _write_ninja_file_and_build_library
    check_compiler_abi_compatibility(compiler)
  File "/home/tomjur/miniconda3/envs/parkour/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 294, in check_compiler_abi_compatibility
    if not check_compiler_ok_for_platform(compiler):
  File "/home/tomjur/miniconda3/envs/parkour/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 254, in check_compiler_ok_for_platform
    which = subprocess.check_output(['which', compiler], stderr=subprocess.STDOUT)
  File "/home/tomjur/miniconda3/envs/parkour/lib/python3.8/subprocess.py", line 415, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/home/tomjur/miniconda3/envs/parkour/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['which', 'c++']' returned non-zero exit status 1.
`

Would it be possible for you to publish your pip list? or maybe an updated install script in case the above is out of date?

Thanks,
Tom

ZiwenZhuang, 2024-07-10:
Yeah, I will update this repo later.

Two options to solve this issue:

1. add `np.float = np.float32` at the start of `train.py`
2. use numpy version of `1.20.1` (hope this will help)

tomjur, 2024-07-10:
Thank you for the quick response!

1. This solution did not help. Probably because the problem is not that the code in `train.py` calls `np.float`, but rather the code somewhere in Isaac calls `np.float` (see callstack above) - so if I understand correctly this is not the relevant scope. Right?
2. Using numpy 1.20.1 did not help - gives the same compilation error (second callstack in the previous message).

ZiwenZhuang, 2024-07-10:
Yeah... You need to `import numpy as np` at the first line of `train.py`. Then, at the second line: `np.float = np.float32`. Then, import isaacgym, torch, rsl_rl, etc.