ZiwenZhuang/parkour #46: Problem while training a walk policy

krank10, 2024-07-11:
Hi,
When I set `selected = "TerrainPerlin"` to train a walk policy, an error occurred：
` File "/home/robot/parkour-main/legged_gym/legged_gym/utils/terrain/perlin.py", line 20, in __init__
    assert(self.xSize == cfg.horizontal_scale * self.tot_rows and self.ySize == cfg.horizontal_scale * self.tot_cols)
AssertionError`.
and when I tried to solve it by adding:
`  self.xSize = cfg.horizontal_scale * self.tot_rows
   self.ySize = cfg.horizontal_scale * self.tot_cols`
before the 'assert' line in `perlin.py`, the error disappear, but the traing process stuck at the beginning.
Would you please solve this problem?

Thanks.

ZiwenZhuang, 2024-07-11:
Could you locate where the program stuck (probably by pressing `Ctrl+C`)? If you have another computationally heavy job running on your server, it could also stuck the initialization of isaacgym.

Xiaodao-chen, 2025-05-18:
I meet the same problem, have you solved it?