ZiwenZhuang/parkour #48: Distillation loss in Nov02_16-18-16_674k_distill_crawljumpjumpleap

Ekaterinador, 2024-07-15:
Hello! I've been testing out a policy that I trained with your repository on the Go2 robot in real life and I've found that only crawl works really well. The good news is that the rest of the skills work in Isaac Gym (except tilt), but unfortunately they don't work in real life. I think the distillation might have been undertrained because I ran it on 32 num_envs instead of the 256 specified in the repository due to graphics card limitations. I've attached a graph of distillation steps 4 and 7 (purple and blue, respectively) below. I'd be really grateful if you could help me figure out what the distillation loss should be at the end of training for a robust launch in real life. 

[image]

meng-zha, 2024-09-03:
Hello, have you solved the problem?
I also want to figure out the distillation loss should be at the end of training. I trained a policy just as the instruction, but the policy don't work well even in the isaac gym.
My distillation loss is here.
[image]

CoderWangcai, 2024-10-09:
Hello, I'm encountering a segmentation fault when executing python legged_gym/scripts/train.py --headless --task go2_distill. Could you please share the environment configuration that worked for you, such as the GPU version?
@meng-zha @Ekaterinador

CoderWangcai, 2024-10-16:
@meng-zha @Ekaterinador 
Hello, have you solved the problem?

Ekaterinador, 2024-10-28:
To be honest, I haven't solved the problem of distilling jumps that can work in real life.