ZiwenZhuang/parkour #49: A1 Code migration

1242713693, 2024-07-16:
Hello, may I ask, is deploying the trained code on A1 similar to the Go1 hardware you released? What do I need to be aware of if I want to port trained code to A1? I just contact with the physical robot, not familiar with, hope to get your reply! Thank you very much

1242713693, 2024-07-16:
> 是的，A1 上的代码与 Go1 上的代码非常相似。但是，我们经历了 A1 电池多次断电吞吐量并损坏了电机。
> 
> 因此，如果您的旧 A1 电池在遇到能量尖峰时可能会欠压，请_**非常小心**_!!

I'm very sorry to bother you. When porting the trained A1 code, do I need to write my own code for A1, like your hardware code for Go1 in the open source project? I'm new to physical robots and don't know much about them! I'm very sorry to bother you, and hope to get your reply, thank you.

ZiwenZhuang, 2024-07-16:
No, you may deploy the hardware code to A1 without much difference. The ROS midware is the same.

1242713693, 2024-07-16:
Thank you very much for your reply. Sorry to bother you