ZiwenZhuang/parkour #54: All scripts but go2 fails on a config issue

fredrikbissmarckatfoise, 2024-08-06:
Hello,
I am an engineer with access to an a1 robot, and experience in RL, so I thought I'd try to apply your impressive parkour package. After having no issues with installation process (I hade to use more modern torch/cuda libs, I have two 4090 cards), all but the go2 application codes fail, it seems to be where the config is supposed to assign defaults to terrain cfg?

```
python legged_gym/scripts/train.py --task a1
Importing module 'gym_38' (/home/frebis/programs/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/frebis/programs/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
PyTorch version 2.4.0+cu118
Device count 2
/home/frebis/programs/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/frebis/.cache/torch_extensions/py38_cu118 as PyTorch extensions root...
Emitting ninja build file /home/frebis/.cache/torch_extensions/py38_cu118/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
Setting seed: 1
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: enabled
Traceback (most recent call last):
  File "legged_gym/scripts/train.py", line 51, in <module>
    train(args)
  File "legged_gym/scripts/train.py", line 45, in train
    env, env_cfg = task_registry.make_env(name=args.task, args=args)
  File "/home/frebis/programs/parkour/legged_gym/legged_gym/utils/task_registry.py", line 99, in make_env
    env = task_class(   cfg=env_cfg,
  File "/home/frebis/programs/parkour/legged_gym/legged_gym/envs/base/legged_robot.py", line 79, in __init__
    super().__init__(self.cfg, sim_params, physics_engine, sim_device, headless)
  File "/home/frebis/programs/parkour/legged_gym/legged_gym/envs/base/base_task.py", line 74, in __init__
    self.create_sim()
  File "/home/frebis/programs/parkour/legged_gym/legged_gym/envs/base/legged_robot.py", line 537, in create_sim
    self._create_terrain()
  File "/home/frebis/programs/parkour/legged_gym/legged_gym/envs/base/legged_robot.py", line 1435, in _create_terrain
    self.terrain = get_terrain_cls(terrain_cls)(self.cfg.terrain, self.num_envs)
  File "/home/frebis/programs/parkour/legged_gym/legged_gym/utils/terrain/__init__.py", line 11, in get_terrain_cls
    entry_point = terrain_registry[terrain_cls]
KeyError: False

```

The value of getattr(self.cfg.terrain, "selected", None) becomes not None, but False. Why may this be?
(I also note that  IsaacGym is no longer supported, would it be straight forward to depend on IsaacLab instead?)

ZiwenZhuang, 2024-08-06:
We updated the code structure in a more unified and extendable way. Some of the configurations are not supported.

Please refer to Go2's configuration as the example.

We will update the configuration examples for the other robot models.

Best,

fredrikbissmarckatfoise, 2024-08-08:
Great, thanks.