ZiwenZhuang/parkour #55: go2 Segmentation fault (core dumped) at distill stage

luoleicn, 2024-08-11:
python legged_gym/scripts/train.py --task go2_distill --sim_device=cuda --headless
segment fault happened while running the following line:
self.sim = self.gym.create_sim(self.sim_device_id, self.graphics_device_id, self.physics_engine, self.sim_params)

sim_params:
```
{
   "sim":{
      "body_measure_points":{
         "base":{
            "x":[
               -0.24,
               -0.21,
               -0.18,
               -0.15,
               -0.12,
               -0.09,
               -0.06,
               -0.03,
               0.0,
               0.030000000000000027,
               0.06,
               0.08999999999999997,
               0.12,
               0.15000000000000002,
               0.18,
               0.20999999999999996,
               0.24,
               0.27,
               0.30000000000000004,
               0.32999999999999996,
               0.36,
               0.39
            ],
            "y":[
               -0.08,
               -0.04,
               0.0,
               0.04,
               0.08
            ],
            "z":[
               -0.061,
               -0.031,
               -0.0010000000000000009,
               0.028999999999999998,
               0.059
            ],
            "transform":[
               0.0,
               0.0,
               0.005,
               0.0,
               0.0,
               0.0
            ]
         },
         "thigh":{
            "x":[
               -0.16,
               -0.158,
               -0.156,
               -0.154,
               -0.152,
               -0.15,
               -0.145,
               -0.14,
               -0.135,
               -0.13,
               -0.125,
               -0.12,
               -0.115,
               -0.11,
               -0.105,
               -0.1,
               -0.095,
               -0.09,
               -0.085,
               -0.08,
               -0.075,
               -0.07,
               -0.065,
               -0.05,
               0.0,
               0.05,
               0.1
            ],
            "y":[
               -0.015,
               -0.01,
               0.0,
               -0.01,
               0.015
            ],
            "z":[
               -0.03,
               -0.015,
               0.0,
               0.015
            ],
            "transform":[
               0.0,
               0.0,
               -0.1,
               0.0,
               1.57079632679,
               0.0
            ]
         },
         "calf":{
            "x":[
               -0.13,
               -0.1,
               -0.07,
               -0.04000000000000001,
               -0.010000000000000009,
               0.01999999999999999,
               0.04999999999999999,
               0.07999999999999999,
               0.10999999999999999
            ],
            "y":[
               -0.015,
               0.0,
               0.015
            ],
            "z":[
               -0.015,
               0.0,
               0.015
            ],
            "transform":[
               0.0,
               0.0,
               -0.11,
               0.0,
               1.57079632679,
               0.0
            ]
         }
      },
      "dt":0.005,
      "gravity":[
         0.0,
         0.0,
         -9.81
      ],
      "no_camera":false,
      "physx":{
         "bounce_threshold_velocity":0.5,
         "contact_collection":2,
         "contact_offset":0.01,
         "default_buffer_size_multiplier":5,
         "max_depenetration_velocity":1.0,
         "max_gpu_contact_pairs":8388608,
         "num_position_iterations":4,
         "num_threads":10,
         "num_velocity_iterations":0,
         "rest_offset":0.0,
         "solver_type":1
      },
      "substeps":1,
      "up_axis":1
   }
}
```

If I set no_camera=True, self.gym.create_sim will success but the system still will fail because of could not find camera with handle

luoleicn, 2024-08-11:
self.sim_device_id is 0, self.graphics_device_id is 0 , self.physics_engine is SimType.SIM_PHYSX

CoderWangcai, 2024-10-10:
I'm deployed to run within a container. When I connect to the container via SSH and execute the command python legged_gym/scripts/train.py --headless --task go2_distill, like you, I encounter a segmentation fault. However, when I connect to the container using TurboVNC and execute the command, there is no error.
[image]

luoleicn, 2024-10-10:
i guess graphic interface is needed, the program will fail on ssh or virtual machine

hamidh1, 2025-05-21:
@CoderWangcai  Hi, Have you resolved this issue? I am getting the same issue.