ZiwenZhuang/parkour #57: Training time and GPU memory of distilllation stage

meng-zha, 2024-08-12:
Thanks for the great work.
I use one 4090 gpu with 24G memory to reproduce the project. While the distillation stage, OOM came out when set "multi_process_" False and use the default config. Since then I reduced the "num_envs" and grid sizes, the training time is extremely long to 30 days. So, I would like to know about:
- the nums_envs, num cols, num rows for 24G 4090
- the normal training time of distillation stage

CoderWangcai, 2024-10-10:
I encountered the same problem as you. Did you manage to solve it?

lamb97, 2024-10-27:
I use one 4090 gpu and use the default config.The training time is long to 50 days.Is this normal?

hanlin-ga, 2024-11-14:
> I use one 4090 gpu and use the default config.The training time is long to 50 days.Is this normal?

I have the same problem currently, is there anyway to run the tensorboard to check the training process?