ZiwenZhuang/parkour #58: Pre-training dataset missing when using go2 for distillation

yue-heu, 2024-08-31:
Thank you for your contribution. I encountered a problem. When I ran the Go2 example, the first and second steps were completed smoothly, but I don’t know how to modify the "{A temporary directory to store collected trajectory}" in the third step. The code location is [go2_distill_config.py L214](https://github.com/ZiwenZhuang/parkour/blob/1ffd6d7c057366da353dd431b5bbd9791f73603f/legged_gym/legged_gym/envs/go2/go2_distill_config.py#L214). Please tell me how to modify it and how to generate this pre-training dataset.
[image]

ZiwenZhuang, 2024-08-31:
Hi,

There is no “dataset” in this pipeline. For running DAgger implementation with multiple processes, this directory is just a temporary place to put the collected trajectories and feed to the trainer process.

Please choose any directly that suits your system and can be read/write fast enough by your program.

Best,

yue-heu, 2024-08-31:
Thank you for your reply. I set a path randomly. When I run the program, it will show "RolloutDataset: trajectory not enough, need 1500 at least, waiting for 15 minutes...". I think there is something wrong with my settings. What problems should I avoid?

ZiwenZhuang, 2024-08-31:
If you have already started the `collect.py` process, just keep waiting.

The prompt in the terminal is just saying there is not enough trajectory in the dataset directory. The process will re-check the directly after 15 minutes.

yue-heu, 2024-08-31:
Thanks for your reply. I tried it, but still can't train in parallel. 
I'm currently training on a 3090 (multi_process_ = False), but it seems too slow. I have multiple 3090 on my server. When I try to execute 
"python legged_gym/scripts/collect.py --headless --task go2_distill --log --load_run "my_log_path" --rl_device cuda:1" 
in another terminal, it still trains on cuda:0, so there will be memory problems. In addition, I tried to set export CUDA_VISIBLE_DEVICES=1 in another terminal, but this will not work directly because the selected gpu is still cuda:0. Do you know how to solve this problem? Thank you very much for your help during the weekend break.

yue-heu, 2024-09-02:
I can run it like this, thank you very much for your contribution of open source code.

CUDA_VISIBLE_DEVICES=0,1 python legged_gym/scripts/collect.py --headless --task go2_distill --log --load_run {the log directory name} --sim_device cuda:1

RoboticsLee, 2024-09-04:
Hello! 
When I run the distillation process of Go2 in multiple processes with multiple GPUs and enough samples were collected, the terminal of `train.py` will have the following error. May I ask if you have encountered this problem?
[image]

yue-heu, 2024-09-06:
> Hello! When I run the distillation process of Go2 in multiple processes with multiple GPUs and enough samples were collected, the terminal of `train.py` will have the following error. May I ask if you have encountered this problem? ![93a00d52503fe3c81eaf2b785223d80](https://private-user-images.githubusercontent.com/91593411/364293229-2cf8cd10-0791-4d00-9548-85b11fc2cd96.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU2MzI5NTAsIm5iZiI6MTcyNTYzMjY1MCwicGF0aCI6Ii85MTU5MzQxMS8zNjQyOTMyMjktMmNmOGNkMTAtMDc5MS00ZDAwLTk1NDgtODViMTFmYzJjZDk2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTA2VDE0MjQxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUzYjY1NDE2ZTAzMGU2YWU3ZGU0MzIyNmUyMzRlZTRlYjZkZWQ5ZmNhZWZhOWRjMzgxYmUzYzE2ZWU0MmI4NzEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Yrxp5SWPAI0ktuhNdNBAdWCzQJR0cOMiQaG2Y3SBb8k)

I also encountered the same problem, but didn't solve it.

ZiwenZhuang, 2024-09-06:
> Hello! When I run the distillation process of Go2 in multiple processes with multiple GPUs and enough samples were collected, the terminal of `train.py` will have the following error. May I ask if [...]

Hi, I've updated the code. Please refer to commit `637080edaf7a04ceee88975d564a578e80c88581` and check if your issue is fixed.

Best,

yue-heu, 2024-09-07:
> Hi, I've updated the code. Please refer to commit `637080edaf7a04ceee88975d564a578e80c88581` and check if your issue is fixed.
>
> Best,

Thanks for the update, it works fine now. I trained it with a single card, but it doesn't look good when I play it in simulation. I trained it with go2, is it my problem? The simulation in the second stage looks good. Is this normal? How do I adjust it?

closed 2024-09-08 by yue-heu

lamb97, 2024-11-04:
> Thanks for the update, it works fine now. I trained it with a single card, but it doesn't look good when I play it in simulation. I trained it with go2, is it my problem? The simulation in the [...]

I have the same problem.Do you solve it?