ZiwenZhuang/parkour #63: Go2 Depth Camera External Pose

CoderWangcai, 2024-10-10:
In the class `sensor-->class forward_camera`, what coordinate system are the `position` and `rotation` in?
On my Go2, the installation position of the D435i is shown in the image below. Is there a significant difference from the default position in the code?

[image]

ZiwenZhuang, 2024-11-04:
Hi,

You probably need to calibrate the camera pose by yourself. The `position` and `rotation` are set w.r.t to the robot base link (as described in the Urdf file)

CoderWangcai, 2024-11-05:
I changed `env_cfg.viewer.draw_sensors` to `True` in play and enabled the `Show axes` in the `base` of IsaacGym, as shown in the picture below.

[image]

I feel that the difference between these two and the position and rotation given in `go2_distill_config.py` do not correspond:
```
            position = dict(
                mean= [0.24, -0.0175, 0.12],
                std= [0.01, 0.0025, 0.03],
            )
            rotation = dict(
                lower= [-0.1, 0.37, -0.1],
                upper= [0.1, 0.43, 0.1],
            )
```
Moreover, when I play and save the last dimension of obs as height_measurements (48x64) to check, I always see the triangular head of Go2, whereas in reality, when I place the D435i at this position, I can never see the triangular head of Go2 regardless of what I do.

[image]

Could you please tell me if there is something I am misunderstanding? Thank you.

manipulability, 2025-02-27:
I’m dealing with the same issue. 
Were you able to find a solution? 
If so, I’d really appreciate it if you could share the camera’s position and rotation.

change-every, 2025-03-21:
> I’m dealing with the same issue. Were you able to find a solution? If so, I’d really appreciate it if you could share the camera’s position and rotation.

Me too, have you found a solution?