ZiwenZhuang/parkour #66: Problem in task Go2_distill

Yan-shiyang, 2024-10-31:
Thanks for authors' source code and excellent work. When I try to train on Unitree Go2, I get the training curve as below. Task go2 and go2_field performs well, but task go2_distill has a problem. The mean_reward is low and has a big variance. And the trained policy plays unwell. Maybe somewhere I made wrong? Anticipating your reply.
[image]
[image]

ZiwenZhuang, 2024-10-31:
The reward curve is OK, since distillation process is completely different from RL. However, your depth figure looks strange.

Does it scale correctly?

Yan-shiyang, 2024-10-31:
Well, I didn't change the scale parameter. I only changed the camera positon and rotation based on deployment situation in go2_distill config. The depth image in the picture is directly  plot using code  ```
 plt.imshow(depth_images[0].cpu().numpy(),cmap= "gray",
                   vmin=self.cfg.sensor.forward_camera.depth_range[0],
                   vmax=self.cfg.sensor.forward_camera.depth_range[1])
```. As you said,it is really strange.

hanlinniu, 2024-11-13:
How long did it cost to train the go2_distill policy? Mine showing it needs 48 days with Nvidia 4090 GPU, is this normal?

hamidh1, 2025-05-20:
@Yan-shiyang  
### Hi, I am unable to train go2_distill policy.  can you please provide support with this. 

Here is the terminal  i am seeing:

```
/home/data3/parkour-main/legged_gym/legged_gym/scripts$ python train.py --headless --task go2_distill 
Importing module 'gym_38' (/home/data3/parkour-main/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/data3/parkour-main/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
PyTorch version 2.4.1+cu121
Device count 2
/home/data3/parkour-main/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/ustc/.cache/torch_extensions/py38_cu121 as PyTorch extensions root...
Emitting ninja build file /home/ustc/.cache/torch_extensions/py38_cu121/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
Setting seed: 1
Using LeggedRobotField.__init__, num_obs and num_privileged_obs will be computed instead of assigned.
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: enabled
heightfield_raw data shape: 1168 528 border size: 200
/home/ustc/anaconda3/envs/pakour/lib/python3.8/site-packages/torch/functional.py:513: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at ../aten/src/ATen/native/TensorShape.cpp:3609.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
Total number of volume estimation points for each robot is: 3034
Actor MLP: Sequential(
  (0): Linear(in_features=256, out_features=512, bias=True)
  (1): ELU(alpha=1.0)
  (2): Linear(in_features=512, out_features=256, bias=True)
  (3): ELU(alpha=1.0)
  (4): Linear(in_features=256, out_features=128, bias=True)
  (5): ELU(alpha=1.0)
  (6): Linear(in_features=128, out_features=12, bias=True)
)
Critic MLP: Sequential(
  (0): Linear(in_features=256, out_features=512, bias=True)
  (1): ELU(alpha=1.0)
  (2): Linear(in_features=512, out_features=256, bias=True)
  (3): ELU(alpha=1.0)
  (4): Linear(in_features=256, out_features=128, bias=True)
  (5): ELU(alpha=1.0)
  (6): Linear(in_features=128, out_features=1, bias=True)
)
Actor RNN: Memory(
  (rnn): GRU(80, 256)
)
Critic RNN: Memory(
  (rnn): GRU(80, 256)
)
Actor MLP: Sequential(
  (0): Linear(in_features=256, out_features=512, bias=True)
  (1): ELU(alpha=1.0)
  (2): Linear(in_features=512, out_features=256, bias=True)
  (3): ELU(alpha=1.0)
  (4): Linear(in_features=256, out_features=128, bias=True)
  (5): ELU(alpha=1.0)
  (6): Linear(in_features=128, out_features=12, bias=True)
)
Critic MLP: Sequential(
  (0): Linear(in_features=256, out_features=512, bias=True)
  (1): ELU(alpha=1.0)
  (2): Linear(in_features=512, out_features=256, bias=True)
  (3): ELU(alpha=1.0)
  (4): Linear(in_features=256, out_features=128, bias=True)
  (5): ELU(alpha=1.0)
  (6): Linear(in_features=128, out_features=1, bias=True)
[... 12 lines cut ...]
  loaded_dict = torch.load(path)
 Warning: using a hacky way to load the model. 
 Replacing encoder.0 weights with untrained weights and avoid critic_encoder.0 
key: encoders.0.conv.conv.0.weight shape: torch.Size([16, 1, 5, 5]) using untrained module weights.
key: encoders.0.conv.conv.0.bias shape: torch.Size([16]) using untrained module weights.
key: encoders.0.conv.conv.3.weight shape: torch.Size([32, 16, 4, 4]) using untrained module weights.
key: encoders.0.conv.conv.3.bias shape: torch.Size([32]) using untrained module weights.
key: encoders.0.conv.conv.6.weight shape: torch.Size([32, 32, 3, 3]) using untrained module weights.
key: encoders.0.conv.conv.6.bias shape: torch.Size([32]) using untrained module weights.
key: encoders.0.head.model.0.weight shape: torch.Size([128, 2464]) using untrained module weights.
key: encoders.0.head.model.0.bias shape: torch.Size([128]) using untrained module weights.
key: encoders.0.head.model.2.weight shape: torch.Size([32, 128]) using untrained module weights.
key: encoders.0.head.model.2.bias shape: torch.Size([32]) using untrained module weights.
 Done: using a hacky way to load the model. 
Initialization done, start learning.
NOTE: you may see a bunch of `NaN or Inf found in input tensor` once and appears in the log. Just ignore it if it does not affect the performance.
RolloutDataset: 0 trajectories found. 0 timesteps in total.
RolloutDataset: trajectory not enough, need 1500 at least, waiting for 15 minutes...
RolloutDataset: 0 trajectories found. 0 timesteps in total.
RolloutDataset: trajectory not enough, need 1500 at least, waiting for 15 minutes...
RolloutDataset: 0 trajectories found. 0 timesteps in total.
RolloutDataset: trajectory not enough, need 1500 at least, waiting for 15 minutes...
RolloutDataset: 0 trajectories found. 0 timesteps in total.
RolloutDataset: trajectory not enough, need 1500 at least, waiting for 15 minutes...
```

hamidh1, 2025-06-04:
@ZiwenZhuang  I have RTX 3060 GPUs. Is the Go2_distill can be done on this and how many days it needs to take the training?

MRMUSASH, 2025-07-09:
> Thanks for authors' source code and excellent work. When I try to train on Unitree Go2, I get the training curve as below. Task go2 and go2_field performs well, but task go2_distill has a problem. [...]

Hello, if possible, I would like to ask how to learn useful strategies more effectively during the distillation phase. At first, my average reward was constantly below 0 (Figure 1). But when I set the trainer forward=0.5 (the collector was still the default 1.0), I found that the average reward began to rise (as shown in Figure 2 for the last 20k of training). Unfortunately, the visual simulation did not work well. @ZiwenZhuang  @Yan-shiyang 

Figure 1:
[image]
Figure 2:
[image]