ZiwenZhuang/parkour #69: Super long training time for go2

hanlinniu, 2024-11-12:
I am running the distillation training for go2 (go2_distill), it shows the ETA time is: 4233451s, it is around 49 days, is this normal?  Is there any trained model we can use directly?

ZiwenZhuang, 2024-11-13:
I set the training time to extremely long only to prevent the program from auto-stopping, since the collector process and trainer process might have different pace on different machines.

hanlinniu, 2024-11-13:
> I set the training time to extremely long only to prevent the program from auto-stopping, since the collector process and trainer process might have different pace on different machines.

Thank you so much for the reply! I am using single RTX 4090 GPU, how long do you think it will take to train the go2_distill policy?

hamidh1, 2025-05-21:
@hanlinniu  
### Hi, I am unable to train go2_distill policy. I am using RTX 3090 GPU (24GB memory) for training. When i try to run the go2_distill policy, I am getting these logs :

_Could you provide any support to resolve this issue?_ 

```
/home/data3/parkour-main/legged_gym/legged_gym/scripts$ python train.py --headless --task go2_distill 
Importing module 'gym_38' (/home/data3/parkour-main/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/data3/parkour-main/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
PyTorch version 2.4.1+cu121
Device count 2
/home/data3/parkour-main/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/ustc/.cache/torch_extensions/py38_cu121 as PyTorch extensions root...
Emitting ninja build file /home/ustc/.cache/torch_extensions/py38_cu121/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
Setting seed: 1
Using LeggedRobotField.__init__, num_obs and num_privileged_obs will be computed instead of assigned.
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: enabled
heightfield_raw data shape: 1168 528 border size: 200
/home/ustc/anaconda3/envs/pakour/lib/python3.8/site-packages/torch/functional.py:513: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at ../aten/src/ATen/native/TensorShape.cpp:3609.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
Total number of volume estimation points for each robot is: 3034
Actor MLP: Sequential(
  (0): Linear(in_features=256, out_features=512, bias=True)
  (1): ELU(alpha=1.0)
  (2): Linear(in_features=512, out_features=256, bias=True)
  (3): ELU(alpha=1.0)
  (4): Linear(in_features=256, out_features=128, bias=True)
  (5): ELU(alpha=1.0)
  (6): Linear(in_features=128, out_features=12, bias=True)
)
Critic MLP: Sequential(
  (0): Linear(in_features=256, out_features=512, bias=True)
  (1): ELU(alpha=1.0)
  (2): Linear(in_features=512, out_features=256, bias=True)
  (3): ELU(alpha=1.0)
  (4): Linear(in_features=256, out_features=128, bias=True)
  (5): ELU(alpha=1.0)
  (6): Linear(in_features=128, out_features=1, bias=True)
)
Actor RNN: Memory(
  (rnn): GRU(80, 256)
)
Critic RNN: Memory(
  (rnn): GRU(80, 256)
)
Actor MLP: Sequential(
  (0): Linear(in_features=256, out_features=512, bias=True)
  (1): ELU(alpha=1.0)
  (2): Linear(in_features=512, out_features=256, bias=True)
  (3): ELU(alpha=1.0)
  (4): Linear(in_features=256, out_features=128, bias=True)
  (5): ELU(alpha=1.0)
  (6): Linear(in_features=128, out_features=12, bias=True)
)
Critic MLP: Sequential(
  (0): Linear(in_features=256, out_features=512, bias=True)
  (1): ELU(alpha=1.0)
  (2): Linear(in_features=512, out_features=256, bias=True)
  (3): ELU(alpha=1.0)
  (4): Linear(in_features=256, out_features=128, bias=True)
  (5): ELU(alpha=1.0)
  (6): Linear(in_features=128, out_features=1, bias=True)
[... 12 lines cut ...]
  loaded_dict = torch.load(path)
 Warning: using a hacky way to load the model. 
 Replacing encoder.0 weights with untrained weights and avoid critic_encoder.0 
key: encoders.0.conv.conv.0.weight shape: torch.Size([16, 1, 5, 5]) using untrained module weights.
key: encoders.0.conv.conv.0.bias shape: torch.Size([16]) using untrained module weights.
key: encoders.0.conv.conv.3.weight shape: torch.Size([32, 16, 4, 4]) using untrained module weights.
key: encoders.0.conv.conv.3.bias shape: torch.Size([32]) using untrained module weights.
key: encoders.0.conv.conv.6.weight shape: torch.Size([32, 32, 3, 3]) using untrained module weights.
key: encoders.0.conv.conv.6.bias shape: torch.Size([32]) using untrained module weights.
key: encoders.0.head.model.0.weight shape: torch.Size([128, 2464]) using untrained module weights.
key: encoders.0.head.model.0.bias shape: torch.Size([128]) using untrained module weights.
key: encoders.0.head.model.2.weight shape: torch.Size([32, 128]) using untrained module weights.
key: encoders.0.head.model.2.bias shape: torch.Size([32]) using untrained module weights.
 Done: using a hacky way to load the model. 
Initialization done, start learning.
NOTE: you may see a bunch of `NaN or Inf found in input tensor` once and appears in the log. Just ignore it if it does not affect the performance.
RolloutDataset: 0 trajectories found. 0 timesteps in total.
RolloutDataset: trajectory not enough, need 1500 at least, waiting for 15 minutes...
RolloutDataset: 0 trajectories found. 0 timesteps in total.
RolloutDataset: trajectory not enough, need 1500 at least, waiting for 15 minutes...
RolloutDataset: 0 trajectories found. 0 timesteps in total.
RolloutDataset: trajectory not enough, need 1500 at least, waiting for 15 minutes...
RolloutDataset: 0 trajectories found. 0 timesteps in total.
RolloutDataset: trajectory not enough, need 1500 at least, waiting for 15 minutes...
```

Crazy-Wxs, 2025-12-26:
> 你好，我这边用单张4090大约需要50天。预计到达时间是4331469.8秒。

您好 请问你训练完了了  我是用单卡4090也需要大概50天   目前已经第四天了  我认为这太长了   请问你有训练的好蒸馏模型吗