ZiwenZhuang/parkour #73: Go2 cannot stand

Gary-Li001, 2024-12-25:
I have completed the deployment and successfully run T-visual and T-run through two terminals. I have also turned off the sport_made service of go2 and entered -- nodryrun in the command line. I controlled it through the remote control and pressed L2+A. However, go2 has been lying down without any movement and cannot stand up.This is my terminal window image:
[image]
[image]

ZiwenZhuang, 2024-12-25:
As I remember, I set L2 or R2 for emergency stop button. Either of the button is hit when the script runs, the script will send a turn-off command message to the system and exit.

Gary-Li001, 2024-12-25:
> 我记得，我将 L2 或 R2 设置为紧急停止按钮。当脚本运行时，点击任一按钮，脚本都会向系统发送关闭命令消息并退出。

Which buttons should I press

deeprail, 2025-01-05:
I met the same problem. 
I pressed a random button and then the commandline told me that Go2 should be run.
But my Go2 just "twitched" a few times and then stayed in a down position, and then did nothing else. 
I'm not quite sure what I'm supposed to do next, either.

Gary-Li001, 2025-01-14:
> 我遇到了同样的问题。我随机按下了一个按钮，然后命令行告诉我应该运行 Go2。但是我的 Go2 只是“抽搐”了几次，然后保持在向下的位置，然后什么也没做。我也不太确定我下一步应该做什么。

My go2 can currently stand up with a twitch, but it can only stand up a little, and it keeps twitching when running, and it can't even climb over very low steps

deeprail, 2025-01-14:
> > 我遇到了同样的问题。我随机按下了一个按钮，然后命令行告诉我应该运行 Go2。但是我的 Go2 只是“抽搐”了几次，然后保持在向下的位置，然后什么也没做。我也不太确定我下一步应该做什么。
> 
> 
> 
> My go2 can currently stand up with a twitch, but it can only stand up a little, and it keeps twitching when running, and it can't even climb over very low steps

I overlook the code and, an option
called "use stand policy" is set to "True". That means that the robot will start with stand policy. 

After the stand policy, you need to click L1 button to stop this policy. (please read the code) Next you will found some more problems, you need to change a device of a tensor. Finally it runs.

But the end result / performance is not good. i am trying to re train my model.

Gary-Li001, 2025-01-14:
> > > 我遇到了同样的问题。 我随机按下了一个按钮，然后命令行告诉我应该运行 Go2。 但是我的 Go2 只是“抽搐”了几次，然后保持在向下的位置，然后什么也没做。 我也不太确定我下一步应该做什么。
> > 
> > 
> > 我的 go2 目前可以通过抽搐站起来，但它只能站起来一点，而且它在跑步时一直抽搐，甚至无法爬过很低的台阶
> 
> 我忽略了代码，并且一个名为“use stand policy”的选项被设置为“True”。这意味着机器人将从 stand policy 开始。
> 
> 在站立策略之后，你需要点击 L1 按钮来停止这个策略。（请阅读代码）接下来你会发现更多的问题，你需要改变一个 tensor 的装置。最后它运行了。
> 
> 但最终结果/性能并不好。我正在尝试重新训练我的模型。

I have modified it and now it can walk normally, but it can't achieve parkour skills. I am trying to modify the pd controller inside.

deeprail, 2025-01-14:
> I have modified it and now it can walk normally, but it can't achieve parkour skills. I am trying to modify the pd controller inside.

I have one more question to discuss, did you find the pose/TF important for training the model?

Gary-Li001, 2025-01-14:
> > > > > 我遇到了同样的问题。 我随机按下了一个按钮，然后命令行告诉我应该运行 Go2。 但是我的 Go2 只是“抽搐”了几次，然后保持在向下的位置，然后什么也没做。 我也不太确定我下一步应该做什么。
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > 我的 go2 目前可以通过抽搐站起来，但它只能站起来一点，而且它在跑步时一直抽搐，甚至无法爬过很低的台阶
> > 
> > 
> > > 
> > 
> > 
> > > 我忽略了代码，并且一个名为“use stand policy”的选项被设置为“True”。这意味着机器人将从 stand policy 开始。
> > 
> > 
> > > 
> > 
> > 
> > > 在站立策略之后，你需要点击 L1 按钮来停止这个策略。 （请阅读代码）接下来你会发现更多的问题，你需要改变一个 tensor 的装置。 最后它运行了。
> > 
> > 
> > > 
> > 
> > 
> > > 但最终结果/性能并不好。 我正在尝试重新训练我的模型。
> > 
> > 
> > 我修改了它，现在它可以正常走路了，但是无法实现跑酷技能。我正在尝试修改里面的 pd 控制器。
> 
> 我还有一个问题要讨论，您觉得姿势/TF 对训练模型很重要吗？

I previously thought that the starting posture had a greater impact on the subsequent movements of go2, but after my experiment (modifying pd, go2 was able to stand up successfully), I found that it still could not walk smoothly, so I think it is a problem with sim2real. I am a novice in this area and I am still looking for a solution.

SABO-6, 2025-03-18:
> I overlook the code and, an option called "use stand policy" is set to "True". That means that the robot will start with stand policy.
>
> After the stand policy, you need to click L1 button to stop [...]

Hi, thank you for sharing your solution and my go2 can also walk following your suggestions. Please, have you retrained the model and deployed it on the robot successfully, so the go2 can complete parkour skills?

sjtu-cjj, 2025-03-25:
[image]
Hello, I am currently trying to deploy my model on Go2, but I encountered the issue shown in the image. I suspect that the problem arises because my ROS environment defaults to Python 3.8, while my virtual environment is using Python 3.6, causing a conflict. My system environment is Ubuntu 20.04, and my ROS version is Foxy. Did you encounter this issue before?

change-every, 2025-03-25:
> [image] Hello, I am currently trying to deploy my model on Go2, but I encountered the issue shown in the image. I suspect that the problem arises because my ROS environment defaults to Python 3.8, [...]

No, i didn't meet this problem. I think you should check your environment and python 3.8 is recommended. 

Btw, have you trained a good distillation policy in the simulation? Maybe we can exchange contact information and help each other.

sjtu-cjj, 2025-03-26:
I trained a distilled model on dual RTX 4090 GPUs for about three days, but the performance is not ideal. It often fails to climb higher steps.

SABO-6, 2025-03-27:
Me too. The mean reward is only about 7 and the performance on Go2 keeps shaking.

I note that the total loss can be separated into distillation and estimation. In addition to depth images, the policy's input also includes the state of the quadrupedal robot. When I set the image to a blank picture, the robot still shakes. So maybe the problem is the estimation?

My loss is 

[image]

I would appreciate it if the author can help us.

lijin255, 2025-04-29:
> > > 我遇到了同样的问题。我随机按下了一个按钮，然后命令行告诉我应该运行 Go2。但是我的 Go2 只是“抽搐”了几次，然后保持在向下的位置，然后什么也没做。我也不太确定我下一步应该做什么。
> > 
> > 
> > My go2 can currently stand up with a twitch, but it can only stand up a little, and it keeps twitching when running, and it can't even climb over very low steps我的 go2 目前能站起来，但只能站起来一点点，跑步的时候一直抽搐，甚至不能爬过很低的台阶
> 
> I overlook the code and, an option我忽略了代码和一个选项 called "use stand policy" is set to "True". That means that the robot will start with stand policy.名为“使用站立策略”的选项设置为“True”。这意味着机器人将以站立策略启动。
> 
> After the stand policy, you need to click L1 button to stop this policy. (please read the code) Next you will found some more problems, you need to change a device of a tensor. Finally it runs.策略执行完成后，需要点击 L1 按钮停止该策略。（请阅读代码）接下来你会发现一些问题，需要更换张量的设备。最终，策略运行成功。
> 
> But the end result / performance is not good. i am trying to re train my model.但最终结果/表现并不好。我正在尝试重新训练我的模型。

请问您训练出来基于go2的一个可用模型了吗？我们能否可以交流一下？

youweicao, 2025-04-29:
Someone (not me) has trained something and I found the repo. [see at https://github.com/hanlinniu/parkour/](https://github.com/hanlinniu/parkour/). I'm not sure how to use it and how it works, what was modified. But I hope it useful for us.
Help each other! I subscribed this issue.

SABO-6, 2025-05-01:
This project appears to be highly challenging to reproduce. I think [Extreme Parkour](https://github.com/chengxuxin/extreme-parkour) seems easier in terms of training and deployment.