ZiwenZhuang/parkour #75: How to Evaluate a Trained Model (from go2feild)？

MRMUSASH, 2025-05-21:
Hello, thank you for your perfect work. 
Because the distillation training took too long (about 40 days), I instead wanted to directly use the model trained by go2feild. I have completed the training and can do visualization. 
However, it is not clear how to evaluate the pass rate of the test model for a single random difficulty obstacle. Could you please give me a guide? Thank you.

hamidh1, 2025-05-21:
@MRMUSASH  ###  Hi, Do you have nay idea to resolve this issue? I am facing this issue during training go2_distill policy:

[image]

MRMUSASH, 2025-06-02:
> [@MRMUSASH](https://github.com/MRMUSASH) ### Hi, Do you have nay idea to resolve this issue? I am facing this issue during training go2_distill policy:
> 
> [image]

i think you need to check your load path?

closed 2025-07-09 by MRMUSASH